OpenNuvoton/NUC980-linux-4.4.y #83: 用buildroot配置系统用户名密码问题，求教，谢谢。

redworlf007, 2022-01-04:
[image]
[image]
[image]

1、我在buildroot里面用make busybox-menuconfig 配置了busybox
2、buildroot里面make menuconfig 配置好，然后make。
3、在串口控制台为啥登录不上去，还要改哪里么？谢谢。

ychuang3, 2022-01-04:
Hi,

在 login: 這裡輸入密碼

YC

redworlf007, 2022-01-04:
> Hi,
> 
> 在 login: 這裡輸入密碼
> 
> YC

你好，是在最后一个图片里面的buildroot login：这里输入密码么？

ychuang3, 2022-01-04:
Hi,

一開始出現的 login:   就是以 root 為 user,  在這裡要輸入 root 的 password
一般 default password 是 "root"

YC

redworlf007, 2022-01-04:
[image]

直接输入密码，不对啊。

schung1218, 2022-01-05:
Hi @redworlf007 ,

buildroot預設是不需要密碼的,  login : root 即可登入,  如果需要密碼, 
你的第一張圖Enable root login with password,  應該就要可以.

請問在沒有編譯過的buildroot, 重新做 
1. make nuvoton_nuc980_xxx_defconfig 
2. 開啟"Enable root login with password"
3. make
也會有同樣的問題嗎?

redworlf007, 2022-01-05:
1、make nuvoton_nuc980_xxx_defconfig，这个默认的是不需要用户名、密码就能登录。
2、開啟"Enable root login with password"，开放这个，设置密码后，同样的问题。

你手上有板子可以试试看。

redworlf007, 2022-01-12:
你好，是试了没，咋样？我搞了好几天还是不行。。。

redworlf007, 2022-01-12:
登录我搞好了，是这样搞得：
1、设置/etc/init.d/rcS文件
添加 /bin/login
2、需要登录：::respawn:/sbin/getty -L ttyS0 115200 vt100

redworlf007, 2022-01-12:
[image]
新的问题，我设置好用户名和密码。
但是系统启动后，如果我再60s内没登录，就等不进去了，这个咋解决？

schung1218, 2022-01-13:
Hi @redworlf007 
根據下列步驟應該可以解決:
1.修改 output/target/etc/inittab
line 34
::once:-/bin/sh
更改為
#::once:-/bin/sh
2. make
NOTE: 如果不需要每次重編都要改話修改 package/busybox/inittab

symfund, 2022-06-02:
一劳永逸地解决办法: 
step 1) 修改：${buildroot}/package/busybox/inittab, 屏蔽掉行：#::once:-/bin/sh

如果你已经编译完系统，不想重新编译，按下面步骤：
step 1) 修改：${buildroot}/package/busybox/inittab, 屏蔽掉行：#::once:-/bin/sh
step 2) make busybox-rebuild
step 3) make

nmm-uto, 2025-02-21:
I am using NUC980  NuMaker-Server-NUC980 and I was trying to enable root login with a password.

For that, I have set 

> System configuration
```
[*] Enable root login with password

(admin) Root password
```

as per @redworlf007 i have set 

[image]

and 

[image]

still board does not prompt for user and password it will directly login with the root

[image]

i have followed @symfund  

and edited [/home/user/NUC970_Buildroot/package/busybox/inittab]

[image]

then 
make busybox-rebuild
make

after this also the board does not ask for a username and password.

I think some files are overriding the buildroot settings.

How can I enable root login with password ?