OpenNuvoton/NUC980-linux-4.4.y #86: nuc980的PG11 PG12 PG13 PG14 PG15被JTAG占用，如果修改为普通io口？

redworlf007, 2022-03-09:
1、nuc980的PG11 PG12 PG13 PG14 PG15被JTAG占用，如果修改为普通io口？谢谢。
2、报错：
Please Check GPIOG13's multi-function = 0x7
sh: write error: Invalid argument

redworlf007, 2022-03-10:
你好，还没有上班么？

redworlf007, 2022-03-10:
江湖告急啊

cwweng-nuvoton, 2022-03-10:
請將 Power On Setting switch 的 PG.4 切為 0
[image]

redworlf007, 2022-03-10:
PA[6:2]已经被用做普通io口了，怎么办？

cwweng-nuvoton, 2022-03-10:
在你的 GPIO 驅動 (drivers/gpio/gpio-nuc980.c) 的 probe(), 將 PG11~15 的 multi-function pin 設為 0

redworlf007, 2022-03-10:
马上改，谢谢。

redworlf007, 2022-03-10:
static int nuc980_gpio_probe(struct platform_device *pdev)
{
	int err;
	struct clk *clk;

#ifndef CONFIG_USE_OF
	if(pdev->id == 0)
#endif
	{
		printk("%s - pdev = %s\n", __func__, pdev->name);
		/* Enable GPIO clock */
		clk = clk_get(NULL, "gpio_hclk");
		if (IS_ERR(clk)) {
			printk(KERN_ERR "nuc980-gpio:failed to get gpio clock source\n");
			err = PTR_ERR(clk);
			return err;
		}
		clk_prepare(clk);
		clk_enable(clk);

		nuc980_gpio_port.dev = &pdev->dev;
		err = gpiochip_add(&nuc980_gpio_port);
		if (err < 0) {
			goto err_nuc980_gpio_port;
		}

	}

#ifdef CONFIG_USE_OF
	{
		struct pinctrl *pinctrl;
		pinctrl = devm_pinctrl_get_select_default(&pdev->dev);
		if (IS_ERR(pinctrl)) {
			return PTR_ERR(pinctrl);
		}
	}
#endif
#ifdef CONFIG_GPIO_NUC980_EINT_WKUP
	nuc980_enable_eint(1,pdev);
#else
	nuc980_enable_eint(0,pdev);
#endif

	return 0;

err_nuc980_gpio_port:
	gpio_ba = 0;
	return err;
}

是在这个函数里面么？谢谢

redworlf007, 2022-03-10:
> 在你的 GPIO 驅動 (drivers/gpio/gpio-nuc980.c) 的 probe(), 將 PG11~15 的 multi-function pin 設為 0

static int nuc980_gpio_probe(struct platform_device *pdev)
{
	int err;
	struct clk *clk;

#ifndef CONFIG_USE_OF
	if(pdev->id == 0)
#endif
	{
		printk("%s - pdev = %s\n", __func__, pdev->name);
		/* Enable GPIO clock */
		clk = clk_get(NULL, "gpio_hclk");
		if (IS_ERR(clk)) {
			printk(KERN_ERR "nuc980-gpio:failed to get gpio clock source\n");
			err = PTR_ERR(clk);
			return err;
		}
		clk_prepare(clk);
		clk_enable(clk);

		nuc980_gpio_port.dev = &pdev->dev;
		err = gpiochip_add(&nuc980_gpio_port);
		if (err < 0) {
			goto err_nuc980_gpio_port;
		}

	}

#ifdef CONFIG_USE_OF
	{
		struct pinctrl *pinctrl;
		pinctrl = devm_pinctrl_get_select_default(&pdev->dev);
		if (IS_ERR(pinctrl)) {
			return PTR_ERR(pinctrl);
		}
	}
#endif
#ifdef CONFIG_GPIO_NUC980_EINT_WKUP
	nuc980_enable_eint(1,pdev);
#else
	nuc980_enable_eint(0,pdev);
#endif

	return 0;

err_nuc980_gpio_port:
	gpio_ba = 0;
	return err;
}

是在这个函数里面么？谢谢。

cwweng-nuvoton, 2022-03-10:
對.

redworlf007, 2022-03-10:
> 對.

具体咋改呢？请教，谢谢。

redworlf007, 2022-03-10:
https://blog.csdn.net/xiaoxilang/article/details/117930676

网上有人这么改了，这个我试了，不行。

cwweng-nuvoton, 2022-03-10:
把下面這一行, 加入 GPIO 驅動 (drivers/gpio/gpio-nuc980.c) 的 probe() 中, 就可以將 PG11~15 的 multi-function pin 設為 0
[image]

修改 GPIO 驅動前, 使用 PG13 會回報錯誤
[image]

修改 GPIO 驅動後, 就可以正常使用 GP13 了
[image]

redworlf007, 2022-03-10:
> 把下面這一行, 加入 GPIO 驅動 (drivers/gpio/gpio-nuc980.c) 的 probe() 中, 就可以將 PG11~15 的 multi-function pin 設為 0 [image]
> 
> 修改 GPIO 驅動前, 使用 PG13 會回報錯誤 [image]
> 
> 修改 GPIO 驅動後, 就可以正常使用 GP13 了 [image]
好的，谢谢了，我改下试试看。

enginlee, 2022-03-11:
分享我的改法：
pin mux會被設為JTAG的根本原因是在u-boot，因為u-boot/board/nuvoton/nuc980/nuc980.c有以下代碼：
`
//Reset multi-function pins to GPIO, except PG[15:11] while JTAGSEL(SYS_PWRON[4]) = 1, or PA[6:2] while JTAGSEL(SYS_PWRON[4]) = 0
	if (readl(REG_PWRON) & 0x00000010)
		writel(0, REG_MFP_GPA_L);
	else
	        writel((readl(REG_MFP_GPA_L) & 0x0FFFFF00), REG_MFP_GPA_L);
...
        if (readl(REG_PWRON) & 0x00000010)
		writel((readl(REG_MFP_GPG_H) & 0xFFFFF000), REG_MFP_GPG_H);
	else
		writel(0, REG_MFP_GPG_H);
`
目的是去讀取外部設置腳位來決定JTAG使用PA[6:2]還是PG[15:11]
只要拿掉if判斷，直接寫入
`
writel(0, REG_MFP_GPA_L);
...
writel(0, REG_MFP_GPG_H);
`
就不須在Linux driver裡面修改了，以上供參考

wosayttn, 2022-03-11:
How about using u-boot mw command and write these commands before jump to kernel?

redworlf007, 2022-03-11:
> How about use u-boot mw command and write these commands into bootcmd variable before jumping to kernel?

环境变量的文件里面可以写uboot 命令。