OpenNuvoton/NUC980-linux-4.4.y #87: 请求添加qualcomm qca7000相关驱动

zgb526642524, 2022-03-15:
> Device Drivers > Network device support > Ethernet driver support
[*]   Qualcomm devices
没有高通qca7000设备的相关驱动

ychuang3, 2022-03-15:
一般我們不會為特定外接設備去開發驅動，
如果是產品規劃需要，請與 Nuvoton FAE 聯繫，提出需求。

zgb526642524, 2022-03-15:
内核从3.18版本开始，已包含QCA7000驱动代码，配置不出qca7000驱动。

ychuang3, 2022-03-15:
config QCA7000
	tristate "Qualcomm Atheros QCA7000 support"
	depends on SPI_MASTER && OF

必須開啟 SPI_MASTER 及設備樹 (device tree) 這個驅動選項才會出現
而且設備樹裡面必須有描述這個裝置，才會在 kernel 被初始化
這些可能需要您自行嘗試

如果是計畫量產的產品而且確認有這個需求，請向 Nuvoton FAE 聯繫

wosayttn, 2022-03-15:
More, please refer the DT binding settings example in Linux documentation.

https://github.com/OpenNuvoton/NUC980-linux-4.4.y/blob/a2cac00fd44fe395f9853afd41c087298d752371/Documentation/devicetree/bindings/net/qca-qca7000-spi.txt