OpenNuvoton/NUC980-linux-4.4.y #88: NUC980 gadget 实现Ethernet + ACM 问题

seawwh, 2022-03-26:
NUC980 gadget 实现Ethernet + ACM 问题

生成 CDC Composite Device (Ethernet and ACM) 图像后下载运行, Win10 上没有反应。
有关情况如下：
1. 单独使用Ethernet 或 ACM win10 图像均可以识别；
2. 使用ifconfig -a 查看usb0 虚拟网卡存在，同时看到 dev/ttyGS0 虚拟串口存在。
3. Win10 设备管理器没有任何反应。

ychuang3, 2022-03-27:
方便提供在 NUC980 的設置 .config 及實現 Ethernet+ACM 的方法嗎?  
我們首先要能複製環境，然後才能查找問題

seawwh, 2022-03-28:
我又做了一个虚拟机unbantu环境下的实验，可以同时出现以太网设备和虚拟串口设备网络。 难道是Win10不支持USB gadget 双设备？

seawwh, 2022-03-28:
这个是在虚拟机ubuntu的截图

seawwh, 2022-03-28:
以下是配置
root@ubuntu:/home/user/NUC970_Buildroot-master# make linux-menuconfig
Device Drivers  ---> 
   [*] USB support  ---> 
   <*>   USB Gadget Support  ---> 
        USB Peripheral Controller  --->     
            <*> NUC980 USB Device Controller                                         
            <*>   USB Gadget Drivers (CDC Composite Device (ACM and mass storage))  --->

seawwh, 2022-03-28:
[image]

seawwh, 2022-03-28:
这是.config 文件
http://www.nuvoton-mcu.com/forum.php?mod=attachment&aid=NDE4Mnw4NDg0ZDhiMnwxNjQ4NDQ3MzU

seawwh, 2022-03-28:
这是.config 文件
http://www.nuvoton-mcu.com/forum.php?od=attachment&aid=NDE4Mnw4NDg0ZDhiMnwxNjQ4NDQ3MzUyfDB8NjgzMzM%3D

seawwh, 2022-03-28:
下面是 .config 文件
[.config.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8360997/default.config.zip)

ychuang3, 2022-03-28:
請參閱這份文件

https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-interface-association-descriptor
"The USB-IF core team has devised a special class and protocol code set that notifies the operating system that one or more IADs are present in device firmware. A device's device descriptor must have the values that appear in the following table or else the operating system will not detect the device's IADs or group the device's interfaces properly."

Windows 要求 composite device 的  Device descriptor 的 bDeviceClass:  bDeviceSubClass : bDeviceProtocol 必須是 0xEF : 0x02 : 0x1
這支 gadget driver 對應的代碼應該是 drivers\usb\gadget\legacy\cdc2.c
裡面的
	.bDeviceClass =		USB_CLASS_COMM,
	.bDeviceSubClass =	0,
	.bDeviceProtocol =	0,
可以嘗試改成 
	.bDeviceClass = 0xEF,
	.bDeviceSubClass =	0x02,
	.bDeviceProtocol =	0x01,
試看看

seawwh, 2022-03-29:
十分感谢您的回复!
经过对cdc2.c 修改后Win10 设备管理器可以出现两个设备一个是串口,一个是 CDC ECM。但是CDC ECM带有黄色问号，网上搜索未见CDC ECM的驱动，可否提供一个CDC ECM驱动？

还有,cdc2.c 同时存在在3个目录中,我修改的是下面第1个,后面的2个需要修改吗?
/home/user/NUC970_Buildroot-master/output/build/linux-master/drivers/usb/gadget/legacy/cdc2.c
/home/user/NUC970_Buildroot-master/output/build/linux-headers-master/drivers/usb/gadget/legacy/cdc2.c
/home/user/NUC970_Buildroot-master/nuc980bsp/linux-4.4.x/drivers/usb/gadget/legacy/cdc2.c

ychuang3, 2022-03-30:
Hi,

我在網絡上搜尋，有好幾個討論串都有提到，Windows 不支援 CDC ECM，而是採用 RNDIS，例如底下兩個連結都有提到

https://social.msdn.microsoft.com/Forums/windowsdesktop/zh-CN/530dd8eb-14c0-4d97-85eb-000f0334d4f7/cdc-ecm-eem?forum=wdk
https://www.kirale.com/support/kb/install-usb-drivers/

seawwh, 2022-03-30:
Hi ychuang3:
我又做了了一个 3合1配置：
  <*>   USB Gadget Drivers (Multifunction Composite Gadget)  --->                    
                    Multifunction Composite Gadget                                              
             [*]       RNDIS + CDC Serial + Storage configuration 
其中网络部分是 RNDIS ，（单独使用RNDIS 网络是OK的）。
运行 Image之后，win10 没有任何反应。

log 输出：

Mass Storage Function, version: 2009/09/11
LUN: removable file: (no medium)
no file given for LUN0
g_multi nuc980-usbdev: failed to start g_multi: -22

附件是详细的控制台输出

[log.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8382012/log.txt)

seawwh, 2022-03-30:
[image]

ychuang3, 2022-03-30:
從 log 訊息 "failed to start g_multi" ，對照原始代碼 ==>  usb device 初始化不成功
原因是 no file given for LUN0
沒有指定 storage 給 mass storage lun0

在網絡上搜尋 "no file given for LUN0" 可以找到一些文章說明