OpenNuvoton/NUC980-linux-4.4.y #89: 請教 NUC980 IoT yaffs2一直掛載不成功

yaojordan, 2022-03-27:
你好,

我想將NUC980 IoT的FS改成yaffs2,
因為不想reset之後就把資料洗掉,
根據許多教學做, 並使用NuWriter燒錄, 雖然成功開機,
但卻還是使用預設的initramfs

這是我的環境
[menuconfig.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8357368/menuconfig.txt)
[kernel_config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8357369/kernel_config.txt)
[env.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8357370/env.txt)

燒錄完成
[image]

開機之後...完全沒有任何關於yaffs的提示

> NUC980 IBR 20180813
> Boot from SPI-NAND
> DDR-OK
> finish SPI dow▒load
> 
>  SPL load main U-Boot from SPI NAND Flash! (Apr  7 2020 14:24:48)
> 
> 
> U-Boot 2016.11 (Apr 07 2020 - 14:24:11 +0800)
> 
> CPU: NUC980
> Board: NUC980
> DRAM:  64 MiB
> NAND:  NAND Flash not found !
> NUC980 NAND CONTROLLER IS NOT SUPPORT THE PAGE SIZE. (0, 0)
> 0 MiB
> SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
> In:    serial
> Out:   serial
> Err:   serial
> Net:   Net Initialization Skipped
> No ethernet found.
> Hit any key to stop autoboot:  0
> SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
> device 0 offset 0x200000, size 0x800000
> SF: 8388608 bytes @ 0x200000 Read: OK
>  Booting kernel from Legacy Image at 00007fc0 ...
>    Image Name:   Linux-4.4.207
>    Image Type:   ARM Linux Kernel Image (uncompressed)
>    Data Size:    5088584 Bytes = 4.9 MiB
>    Load Address: 00008000
>    Entry Point:  00008000
>    Verifying Checksum ... OK
>    XIP Kernel Image ... OK
> 
> Starting kernel ...
> 
> SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
> SF: DeviceBooting Linux on physical CPU 0x0
> Linux version 4.4.207 (user@ubuntu) (gcc version 4.8.5 (Buildroot 2016.11.1-g66ba911) ) #6 PREEMPT Sun Mar 27 17:56:12 CST 2022
> CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
> CPU: VIVT data cache, VIVT instruction cache
> Machine: NUC980
> Memory policy: Data cache writeback
> Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
> Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0(u-boot),0x1400000@0x200000(kernel),-(user)ignore_loglevel
> PID hash table entries: 256 (order: -2, 1024 bytes)
> Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
> Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
> Memory: 56708K/65536K available (4034K kernel code, 288K rwdata, 1252K rodata, 2376K init, 219K bss, 8828K reserved, 0K cma-reserved)
> Virtual kernel memory layout:
>     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
>     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[... 188 lines cut ...]
>  r6:c3b03000 r5:c3b03000 r4:c04d2cdc r3:00000000
> [<c0013014>] (show_stack) from [<c017e298>] (dump_stack+0x20/0x28)
> [<c017e278>] (dump_stack) from [<c0075f28>] (panic+0xb0/0x240)
> [<c0075e7c>] (panic) from [<c054820c>] (mount_block_root+0x224/0x2d8)
>  r3:00000001 r2:00000000 r1:c383fec0 r0:c04d2cdc
>  r7:c3fed060
> [<c0547fe8>] (mount_block_root) from [<c0548424>] (mount_root+0x68/0x84)
>  r10:0000007a r9:c0568838 r8:c056ebd4 r7:c0568830 r6:c0568858 r5:c0568858
>  r4:00000000
> [<c05483bc>] (mount_root) from [<c05485a0>] (prepare_namespace+0x160/0x1c0)
>  r4:c05b8264 r3:00000000
> [<c0548440>] (prepare_namespace) from [<c0547e34>] (kernel_init_freeable+0x184/0x1cc)
>  r6:c05b8240 r5:c05b8240 r4:00000007
> [<c0547cb0>] (kernel_init_freeable) from [<c03ff12c>] (kernel_init+0x10/0xf4)
>  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03ff11c
>  r4:00000000
> [<c03ff11c>] (kernel_init) from [<c000fd78>] (ret_from_fork+0x14/0x3c)
>  r4:00000000 r3:ffffffff
> ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
> random: nonblocking pool is initialized

覺得很怪...參數應該都有設定
bootargs也有吃到
請幫忙看看
萬分感謝

ychuang3, 2022-03-29:
"NUC980 Linux 4.4 BSP User Manual EN"  文件裡面有說明

yaojordan, 2022-03-29:
Dear ychuang,

折騰了半天原來是kernel config沒開FMI(Flash Memory Interface)...
做以下設定後即可:
```
Device Drivers ---> 
    Generic Driver Options --->
        <*> Nuvoton NUC980 FMI function selection
            Select FMI device to support (Support MTD NAND Flash) --->
```
接著將Initial RAM filesystem disabled, 板子reset後便能掛上YAFFS2
```
General setup --->
    [ ] Initial RAM filesystem and RAM disk (initramfs/initrd) support
```

資料查了半天都沒想到去讀官方文件, 非常感謝您一針見血的提點

closed 2022-03-29 by yaojordan