OpenNuvoton/NUC980-linux-4.4.y #90: NUC980 - ETH2UART Board SPI CS0 Support

RSTurgay, 2022-04-13:
Hello, I have nuvoton eth2 uart kit. I am working with SPI NAND flash on board. I just want to use the SPI0 pins with spidev as an extra.

SPI0-CS0 (PD8)
SPI0-DI (PD11)
SPI0-DO (PD10)
SPI0-CLK (PD9)

Here are my Kernel SPI Configuration settings.

[image]

[image]

**I can't use the CS0 Pin. Can I get support from you for this?**

[image]

**No response on CS0 Pin. It's normal not to be because it shows as spidev 1.1 on the system. How can I enable it as spidev1.0.**

[image]

 **I have the following edits facts to add spidev 1.0**

```
#ifdef CONFIG_SPI_SPIDEV
	{
		.modalias = "spidev",
		.max_speed_hz = 75000000,
		.bus_num = 2,
		#if defined(CONFIG_BOARD_IOT) || defined(CONFIG_BOARD_LORAG)
		.chip_select = 0,	//use SS0
		#else
		.chip_select = 1,	//use SS1
		#endif
		.mode = SPI_MODE_0,
	},
#endif
```
**I edited as** 
```
#ifdef CONFIG_SPI_SPIDEV
	{
		.modalias = "spidev",
		.max_speed_hz = 75000000,
		.bus_num = 2,
		#if defined(CONFIG_BOARD_IOT) || defined(CONFIG_BOARD_LORAG)
		.chip_select = 0,	//use SS0
		#else
		.chip_select = 0,	//use SS0
		#endif
		.mode = SPI_MODE_0,
	},
#endif
```

**But I am facing an error on the kernel side. Can you help?**

```
m25p80 spi1.0: unrecognized JEDEC id bytes: ff, ff, ff
m25p80: probe of spi1.0 failed with error -2
```

cwweng-nuvoton, 2022-04-14:
Hi,

Since you'd like to use SPI0 to access SPI NAND, you have to enable SPI0 and **not** to enable "User mode SPI device driver support".
[image]

Maybe you miss following two SPI NAND configurations. Please check.
1. Enable Staging drivers
[image]

2. Enable following two SPI NAND supports
[image]

cwweng-nuvoton, 2022-04-14:
Hi,

I see your error message.
[image]
You use "m25p80" driver to access SPI NAND ?
That's wrong configuration. Since m25p80 driver is for SPI NOR Flash.

If you didn't use QSPI0 to access SPI NOR Flash, remember to disable "SPI-NOR device support" in "Device Drivers"->"Memory Technology Device (MTYD) support".

RSTurgay, 2022-04-14:
Hi, 

Firstly, thanks for your response.

> Since you'd like to use SPI0 to access SPI NAND, you have to enable SPI0 and **not** to enable "User mode SPI device driver support".

I looked at the features you mentioned, they are all enabled but I don't want to use SPI0 to access any NAND Flash. I want to use it to drive an external device.

> Hi,
> 
> I see your error message. [image] You use "m25p80" driver to access SPI NAND ? That's wrong configuration. Since m25p80 driver is for SPI NOR Flash.
> 
> If you didn't use QSPI0 to access SPI NOR Flash, remember to disable "SPI-NOR device support" in "Device Drivers"->"Memory Technology Device (MTD) support".

QSPI0 is used to access the SPI NAND Flash on the board, not SPI NOR.

`If you didn't use QSPI0 to access SPI NOR Flash, remember to disable "SPI-NOR device support" in "Device Drivers"->"Memory Technology Device (MTD) support".`

I disabled this feature you mentioned.

RSTurgay, 2022-04-14:
Also,

When I configure the SPI0-CS1 (PG15) pin, I can use it.

[image]

My Configuration
[image]

The only problem here is that I cannot use the SPI0-CS0(PD8) pin on the kernel. Driver not loading as spidev1.0 in kernel. I am getting the error you mentioned.

[image]

cwweng-nuvoton, 2022-04-14:
Hi,

OK.  I got it.
Do you access SPI device by SPI0 successfully now ?

cwweng-nuvoton, 2022-04-14:
Hi,

From your /dev/
[image]
You set .chip_select to 1

Please modify arch/arm/mach-nuc980/dev.c
Set chip_select to 0
[image]

Then spidev1.0 will be generated in your /dev/

cwweng-nuvoton, 2022-04-14:
spidev**x.y**

x is bus_num
y is chip_select

RSTurgay, 2022-04-14:
> Hi,
> 
> OK. I got it. Do you access SPI device by SPI0 successfully now ?

I am performing the compilation process and I will share the result with you.

RSTurgay, 2022-04-14:
> spidev**x.y**
> 
> x is bus_num y is chip_select

Yes, I know this is so.

Thank you for the information.

RSTurgay, 2022-04-14:
> Hi,
> 
> From your /dev/ [image] You set .chip_select to 1
> 
> Please modify arch/arm/mach-nuc980/dev.c Set chip_select to 0 [image]
> 
> Then spidev1.0 will be generated in your /dev/

Thank you very much indeed for your interest. You can find this process in the first description,  https://github.com/OpenNuvoton/NUC980-linux-4.4.y/issues/90#issue-1202871792

I did this process, but it didn't work. But now I'm recompiling all packages and will try again. And I will share the result with you.

cwweng-nuvoton, 2022-04-14:
From your modification
[image]
It seems you modify to spi1, rather spi0.

RSTurgay, 2022-04-14:
> It seems you modify to spi1, rather spi0.

My fault. I did this when it appeared as spi1.x in the kernel. But now I will do what you mentioned.

Thank you very much for your interest.

cwweng-nuvoton, 2022-04-15:
Hi,

The Kernel panic issue occurred while mounting root file system.
Did you mount root file system on SPI NAND via QSPI0 ?
Which file system you select?  YAFFS2 ?
Could you provide your Kernel .config and full Kernel boot log?

The second issue that the probe() of m25p80 driver sends "read SPI Flash ID command" (0x9F) to SPI device.
But the SPI device cannot recognize the 0x9F command and didn't response it.
Hence, spi0 got ff, ff, ff.

If you didn't use SPI NOR Flash, you can disable "SPI-NOR device support" in "Device Drivers"->"Memory Technology Device (MTD) support".

RSTurgay, 2022-04-15:
Hi,

`Did you mount root file system on SPI NAND via QSPI0 ?` --> Yes

`Which file system you select? YAFFS2 ?` --> Yes I'm using YAFFS2

The file you want. But SPI NOR Flash is not disabled.

[linux-master-config.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8494190/linux-master-config.zip)

`If you didn't use SPI NOR Flash, you can disable "SPI-NOR device support" in "Device Drivers"->"Memory Technology Device (MTD) support".`

If I do this suggestion you mentioned, I get an error like this.

```
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
4-bit
finish SPI dow▒c{
                 #S
 SPL load main U-Boot from SPI NAND Flash! (Apr 03 2022 12:04:58)

U-Boot 2016.11 (Apr 13 2022 - 13:50:33 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  0 MiB
jedec aa21SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x200000, size 0x1400000
 20971520 bytes read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.289
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2270104 Bytes = 2.2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

jedec aa21SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: Device▒Booting Linux on physical CPU 0x0
Linux version 4.4.289 (turgay@turgay-VirtualBox) (gcc version 4.8.5 (Buildroot 2016.11.1-g7116cea) ) #6 PREEMPT Wed Apr 13 15:09:30 +03 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
Memory policy: Data cache writeback
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat c0488588, node_mem_map c3f77000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16384 pages, LIFO batch:3
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[... 121 lines cut ...]
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.289 #6
Hardware name: NUC980
Backtrace:
[<c0012de0>] (dump_backtrace) from [<c0012fcc>] (show_stack+0x18/0x1c)
 r6:c3a18000 r5:c3a18000 r4:c03db084 r3:00000000
[<c0012fb4>] (show_stack) from [<c0344268>] (dump_stack+0x20/0x28)
[<c0344248>] (dump_stack) from [<c0342620>] (panic+0xb0/0x240)
[<c0342574>] (panic) from [<c0434230>] (mount_block_root+0x224/0x2d8)
 r3:00000001 r2:00000000 r1:c383fec0 r0:c03db084
 r7:c3feb300
[<c043400c>] (mount_block_root) from [<c0434448>] (mount_root+0x68/0x84)
 r10:0000006f r9:c044e838 r8:c0454474 r7:c044e830 r6:c044e858 r5:c044e858
 r4:00000000
[<c04343e0>] (mount_root) from [<c04345c4>] (prepare_namespace+0x160/0x1c0)
 r4:c0488da4 r3:00000000
[<c0434464>] (prepare_namespace) from [<c0433e58>] (kernel_init_freeable+0x1a8/0x1f0)
 r6:c0488d80 r5:c0488d80 r4:00000007
[<c0433cb0>] (kernel_init_freeable) from [<c0345654>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0345644
 r4:00000000
[<c0345644>] (kernel_init) from [<c000fd98>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

cwweng-nuvoton, 2022-04-15:
Please disable CONFIG_SPI_NUC980_QSPI0_SS1/CONFIG_SPI_NUC980_QSPI0_SS1_PD0 
You use QSPI0_SS**1**(PD0) for SPI NAND.
However, the SPI NAND on ETH2UART Board is connected with QSPI0_SS**0**

RSTurgay, 2022-04-15:
Hi,

My configuration,

[image]
[image]

I'm change this.
```
#ifdef CONFIG_SPI_SPIDEV
	{
		.modalias = "spidev",
		.max_speed_hz = 75000000,
		.bus_num = 1,
		#if defined(CONFIG_BOARD_IOT) || defined(CONFIG_BOARD_LORAG)
		.chip_select = 0,	//use SS0
		#else
		.chip_select = 0,	//use SS0
		#endif
		.mode = SPI_MODE_0,
	},
#endif
};

```

**But same error !**

```
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
4-bit
finish SPI dow▒c{
                 #S
 SPL load main U-Boot from SPI NAND Flash! (Apr 03 2022 12:04:58)

U-Boot 2016.11 (Apr 13 2022 - 18:05:25 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  0 MiB
jedec aa21SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x200000, size 0x1400000
 20971520 bytes read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.289
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2266352 Bytes = 2.2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

jedec aa21SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: Device▒Booting Linux on physical CPU 0x0
Linux version 4.4.289 (turgay@turgay-VirtualBox) (gcc version 4.8.5 (Buildroot 2016.11.1-g7116cea) ) #4 PREEMPT Thu Apr 14 11:18:14 +03 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
[... 135 lines cut ...]
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.4.289 #4
Hardware name: NUC980
Backtrace:
[<c0012de0>] (dump_backtrace) from [<c0012fcc>] (show_stack+0x18/0x1c)
 r6:c3a10000 r5:c3a10000 r4:c03d8124 r3:00000000
[<c0012fb4>] (show_stack) from [<c0342d5c>] (dump_stack+0x20/0x28)
[<c0342d3c>] (dump_stack) from [<c0341114>] (panic+0xb0/0x240)
[<c0341068>] (panic) from [<c0430230>] (mount_block_root+0x224/0x2d8)
 r3:00000001 r2:00000000 r1:c383fec0 r0:c03d8124
 r7:c3feb200
[<c043000c>] (mount_block_root) from [<c0430448>] (mount_root+0x68/0x84)
 r10:0000006f r9:c044a838 r8:c0450474 r7:c044a830 r6:c044a858 r5:c044a858
 r4:00000000
[<c04303e0>] (mount_root) from [<c04305c4>] (prepare_namespace+0x160/0x1c0)
 r4:c0484da4 r3:00000000
[<c0430464>] (prepare_namespace) from [<c042fe58>] (kernel_init_freeable+0x1a8/0x1f0)
 r6:c0484d80 r5:c0484d80 r4:00000007
[<c042fcb0>] (kernel_init_freeable) from [<c0344148>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0344138
 r4:00000000
[<c0344138>] (kernel_init) from [<c000fd98>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

RSTurgay, 2022-04-15:
> Please disable CONFIG_SPI_NUC980_QSPI0_SS1/CONFIG_SPI_NUC980_QSPI0_SS1_PD0 You use QSPI0_SS**1**(PD0) for SPI NAND. However, the SPI NAND on ETH2UART Board is connected with QSPI0_SS**0**

I didn't pay much attention to that. I did not want to change the default configuration settings to avoid errors.

RSTurgay, 2022-04-15:
Also my U-Boot Environment : 

baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
ipaddr=169.254.131.100
loadkernel=nand read 7fc0 200000 1400000
bootcmd=run loadkernel;bootm 7fc0;
bootargs=noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel

cwweng-nuvoton, 2022-04-15:
Attached are my Kernel config, kernel image and boot log.
[ETH2UART.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8495498/ETH2UART.zip)

Kernel mounts YAFFS2 file system on SPI NAND successfully.
[image]

RSTurgay, 2022-04-15:
Hi, 

Thank you very much for your interest. I'll try to set it up in my own kernel.

closed 2022-04-15 by RSTurgay