OpenNuvoton/NUC980-linux-4.4.y #91: 980 Boot Error Info  “ Bad eraseblock”

seawwh, 2022-04-14:
启动过程中出现的错误信息：
“ Bad eraseblock” 如何解决？

usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
nand: Winbond W25N01GV 1G 3.3V
nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
nand: WARNING: spi0.0: the ECC used on your system is too weak compared to the one required by the NAND chip
Scanning device for bad blocks
Bad eraseblock 5 at 0x0000000a0000
Bad eraseblock 6 at 0x0000000c0000
Bad eraseblock 7 at 0x0000000e0000
Bad eraseblock 10 at 0x000000140000
Bad eraseblock 11 at 0x000000160000
Bad eraseblock 12 at 0x000000180000
Bad eraseblock 13 at 0x0000001a0000
Bad eraseblock 14 at 0x0000001c0000
Bad eraseblock 15 at 0x0000001e0000
usb-storage 1-2:1.0: USB Mass Storage device detected
scsi host0: usb-storage 1-2:1.0
Creating 3 MTD partitions on "nand0":
0x000000000000-0x000000200000 : "u-boot"
0x000000200000-0x000001600000 : "Kernel"
0x000001600000-0x000008000000 : "user"

cwweng-nuvoton, 2022-04-14:
Hi,

SPI NAND 驅動存取 SPI NAND 遇到壞塊時, 會回報給 Linux MTD. 後續 Linux MTD 會跳過壞塊, 存取後面的好塊.
因此針對這個壞塊的錯誤訊息, 使用者不需要另外去解決.

seawwh, 2022-04-15:
https://gitee.com/OpenNuvoton/NUC980-linux-4.4.y/issues/I1395D?from=project-iss
这里有人说是需要加上拉电阻的原因：
[image]

seawwh, 2022-04-15:
我查了一下 NK-980IOT  电路图，其中是有 10K 上拉的
[image]