OpenNuvoton/NUC980-linux-4.4.y #93: 请教NUC980外挂RTC芯片RX-8025T内核配置问题。

redworlf007, 2022-04-15:
1、NUC980外挂RTC芯片RX-8025T，挂在i2c0  PE12 PE10上，内核也配置了，但是系统跑起来，认不到8025。

scottzhang, 2022-04-15:
I2C通信地址的问题吧，你内核打印log看看，我们外挂过其他型号的RTC，你这个没用过

mjchen1, 2022-04-15:
slave mode要拿掉

[image]

redworlf007, 2022-04-15:
好，我马上试试。

redworlf007, 2022-04-15:
U-Boot 2016.11 (Dec 24 2021 - 18:19:14 +0600)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

Loading from nand0, offset 0x200000
   Image Name:   Linux-4.4.179
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2748224 Bytes = 2.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.179
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2748224 Bytes = 2.6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starti▒Booting Linux on physical CPU 0x0
Linux version 4.4.179 (root@zxg-virtual-machine) (gcc version 4.9.4 (Buildroot 2016.11.1) ) #49 PREEMPT Fri Apr 15 12:10:49 +06 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 59100K/65536K available (4078K kernel code, 215K rwdata, 1132K rodata, 140K init, 211K bss, 6436K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc051ef1c   (5212 kB)
      .init : 0xc051f000 - 0xc0542000   ( 140 kB)
      .data : 0xc0542000 - 0xc0577ce0   ( 216 kB)
       .bss : 0xc0577ce0 - 0xc05acaec   ( 212 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8400 - 0x843c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
[... 85 lines cut ...]
yaffs: passed flags "inband-tags"
usb 1-2: new high-speed USB device number 2 using nuc980-ehci
cdc_ether 1-2:1.0 usb0: register 'cdc_ether' at usb-nuc980-ehci-2, CDC Ethernet Device, ae:0c:29:a3:9b:6d
option 1-2:1.2: GSM modem (1-port) converter detected
usb 1-2: GSM modem (1-port) converter now attached to ttyUSB0
option 1-2:1.3: GSM modem (1-port) converter detected
usb 1-2: GSM modem (1-port) converter now attached to ttyUSB1
option 1-2:1.4: GSM modem (1-port) converter detected
usb 1-2: GSM modem (1-port) converter now attached to ttyUSB2
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
devtmpfs: mounted
Freeing unused kernel memory: 140K
Starting logging: OK
Starting mdev...
Initializing random number generator... random: dd: uninitialized urandom read (512 bytes read, 19 bits of entropy available)
done.
Starting rpcbind: OK
Starting network: OK
Starting ntpd: OK
random: ssh-keygen: uninitialized urandom read (32 bytes read, 21 bits of entropy available)
Starting sshd: random: sshd: uninitialized urandom read (32 bytes read, 21 bits of entropy available)
OK
nuc980-emac0 nuc980-emac0: eth0 is OPENED
Starting install openGW ...
Starting Run openGWStart.sh

redworlf007, 2022-04-15:
[image]

redworlf007, 2022-04-15:
i2c /dev entries driver
nuc980-i2c0 nuc980-i2c0: i2c-0: nuc980 I2C adapter

系统启动是这样打印的，然后rx8025还是没有挂上。

https://blog.csdn.net/chenliang0224/article/details/51298682

你看看这个帖子，972上是这样的。

redworlf007, 2022-04-15:
我在这里/home/zxg/work/nuc980bsp/linux-4.4.x/arch/arm/mach-nuc980/dev.c
static struct i2c_board_info __initdata nuc980_i2c_clients0[] = {
#ifdef CONFIG_SND_SOC_NAU8822
	{I2C_BOARD_INFO("nau8822", 0x1a),},
#endif
        {I2C_BOARD_INFO("rx8025", 0x32),},    //添加了这个
};

系统启动后，一直在打印：
rtc-rx8025 0-0032: power voltage drop detected
rtc-rx8025 0-0032: power-on reset detected, date is invalid

redworlf007, 2022-04-15:
用这个命令：i2cdetect 0
[image]

返回32地址，但是系统没有识别到rx8025。

redworlf007, 2022-04-18:
大神，在么？帮忙看看这个是驱动哪里有问题，谢谢了。

wosayttn, 2022-04-18:
Hi,
Did you pull-up both I2C_SDA and SCK pins with resistors? Maybe you should have a investigation using logic analyzer.

redworlf007, 2022-04-18:
> Hi, Did you pull-up both I2C_SDA and SCK pins with resistors? Maybe you should have a investigation using logic analyzer.

两个4.7K的上拉电阻，我用示波器测了下，sck和sda上有波形，但我还没有具体看是啥数据。

redworlf007, 2022-04-18:
[image]

看看，我用i2c-tool，读地址0x32器件的寄存0 1 2 可以读回来数据。

redworlf007, 2022-04-19:
这个器件地址是0x32，读 0寄存器
i2cget -f -y 0 0x32 0

[image]

逻辑分析仪抓的数据，rx8025有应答，i2c通讯成功，但是Linux驱动不识别这个芯片。
我在内核配置上已经选中了，帮忙看看啥问题？谢谢了。

redworlf007, 2022-04-19:
> Hi, Did you pull-up both I2C_SDA and SCK pins with resistors? Maybe you should have a investigation using logic analyzer.

你好，用逻辑分析仪的结果见楼上。

redworlf007, 2022-04-19:
我感觉是980的Linux驱动哪里有问题，就没有挂载rx8025

redworlf007, 2022-04-19:
/home/zxg/work/nuc980bsp/linux-4.4.x/arch/arm/mach-nuc980/dev.c
我在这里这样改了：
/* I2C */
#if defined(CONFIG_I2C_BUS_NUC980_P0) || defined(CONFIG_I2C_BUS_NUC980_P0_MODULE)
// port 0
/* I2C clients */
static struct i2c_board_info __initdata nuc980_i2c_clients0[] = {
#ifdef CONFIG_SND_SOC_NAU8822
	{I2C_BOARD_INFO("nau8822", 0x1a),},
#endif
        {I2C_BOARD_INFO("rx8025", 0x32),},
};

系统启动后
[image]

内核崩溃了。

redworlf007, 2022-04-19:
现在最新的测试结果，RX8025的I2C是通讯成功的，RX8025的FOUT脚上输出32.768的方波，说要芯片硬件工作正常。

目前这个内核奔溃的问题，已改是驱动程序里面指针越界，但是还没有定位到是哪里的问题。

redworlf007, 2022-04-19:
/home/zxg/work/nuc980bsp/linux-4.4.x/arch/arm/mach-nuc980/dev.c

我发现在我修改之前，dev.c里面，压根就没有{I2C_BOARD_INFO("rx8025", 0x32),}, 这句，那么系统怎么找到rx8025的驱动？这里应该有问题。

redworlf007, 2022-04-19:
各位大神，能不能帮忙给看看，到底哪里的问题？我怀疑是驱动的问题。。。。

redworlf007, 2022-04-19:
[image]

我感觉要改设备树，那个方框里面是我加的，好像还没改对。。。

redworlf007, 2022-04-20:
这个问题我搞好了，我用的内核版本4.4.179的版i2c驱动有bug，我升级到最新的内核版本289，问题解决了。