OpenNuvoton/NUC980-linux-4.4.y #94: Kernel panic - not syncing: Out of memory and no killable processes

shawnhsiao97, 2022-04-15:
Hi guys,
The kernel panic while loading kernel at boot if a uImage size larger than 12MB
I confused what a little size was cause OOM
because I thought the DDR size is 64 MB it has should enough memory space
I follow document flash uboot and kernel into qspiNAND flash and then  read kernel from qspi flash to DDR
```
[ pid ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Kernel panic - not syncing: Out of memory and no killable processes
```

Dose anybody know what cause it and can I just boot kernel in DDR and increase the uImage size?

ychuang3, 2022-04-16:
Which type of board are you using and select which defconfig?

shawnhsiao97, 2022-04-18:
> Which type of board are you using and select which defconfig?

Hi @ychuang3 thank you for reply

I used NUC980DK61YC and No I didn't used nuc980_iot_defconfig Instead used need of manuconfig for myself

supplementary log

```
 NUC980 IBR 201SF: 38797312 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    16560776 Bytes = 15.8 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.207+ (r
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow▒load

 SPL load main U-Boot from SPI NAND Flash! (Mar 28 2022 10:33:25)

U-Boot 2016.11-gbf0ff9f-dirty (Mar 28 2022 - 10:33:25 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  NAND Flash not found !
NUC980 NAND CONTROLLER IS NOT SUPPORT THE PAGE SIZE. (0, 0)
0 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x2500000
SF: 38797312 bytes @ 0x200000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.207+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    16560776 Bytes = 15.8 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: DeviceBooting Linux on physical CPU 0x0
Linux version 4.4.207+ (root@shawn-Default-string) (gcc version 11.2.0 (Ubuntu 11.2.0-5ubuntu1) ) #25 PREEMPT Tue Apr 12 16:51:00 CST 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
[... 173 lines cut ...]
[<c0526360>] (write_buffer) from [<c05263e0>] (flush_buffer+0x3c/0xa0)
 r5:00008000 r4:c3990000
[<c05263a4>] (flush_buffer) from [<c0534958>] (__gunzip+0x2b4/0x378)
 r9:c0534690 r8:c0526168 r7:c05263a4 r6:c3990000 r5:c054bae4 r4:00008000
[<c05346a4>] (__gunzip) from [<c0534a50>] (gunzip+0x34/0x3c)
 r10:c04b188c r9:00000000 r8:00000000 r7:00000000 r6:00d4c29c r5:c054bae4
 r4:c05448b8
[<c0534a1c>] (gunzip) from [<c0526724>] (unpack_to_rootfs+0x178/0x2b4)
[<c05265ac>] (unpack_to_rootfs) from [<c0527048>] (populate_rootfs+0x20/0x24c)
 r10:c1297d80 r9:c0521dd8 r8:00000000 r7:c0527028 r6:c129ba40 r5:c39235a0
 r4:c129ba40
[<c0527028>] (populate_rootfs) from [<c0009574>] (do_one_initcall+0x88/0x1fc)
 r10:c05244e8 r9:c0521dd8 r8:00000000 r7:c0527028 r6:c129ba40 r5:c39235a0
 r4:c129ba40
[<c00094ec>] (do_one_initcall) from [<c0524de8>] (kernel_init_freeable+0x10c/0x1c8)
 r9:c0521dd8 r8:c054482c r7:c054484c r6:c12df180 r5:0000007b r4:c054b84c
[<c0524cdc>] (kernel_init_freeable) from [<c03e0cec>] (kernel_init+0x10/0xe8)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03e0cdc
 r4:00000000
[<c03e0cdc>] (kernel_init) from [<c000f6f8>] (ret_from_fork+0x14/0x3c)
 r5:c03e0cdc r4:00000000
---[ end Kernel panic - not syncing: Out of memory and no killable processes...

random: nonblocking pool is initialized
```

ychuang3, 2022-04-18:
Please make kernel with the NUC980 defconfig.

shawnhsiao97, 2022-04-18:
> Please make kernel with the NUC980 defconfig.

Hi @ychuang3 , I have made kernel with the NUC980 defonfig now.
but unfortunately still the same

ychuang3, 2022-04-21:
(1) git clone git@github.com:OpenNuvoton/NUC970_Buildroot.git
(2) make nuvoton_nuc980_iot_defconfig
(3) make
(4) use NuWriter to download output/images/Image
Can it work?

shawnhsiao97, 2022-04-22:
Hi @ychuang3 thank you for reply

Yes, it can work.
but my problem is what uImage size over 12MB will cause kernel panic
AFAIK I think the EVB have 64MB of memory in a DDR, therefore it shouldn't happen

Follow log mentioned above, the reason for the kernel panic is what it cannot allocate the memory when rootfs decompressed by kernel I think.

May be there is caused by memory allocation, but I do not have any idea solve it, do you have it?

ychuang3, 2022-04-22:
My kernel image size is 12 MB, it works well.
Would you upload your .config?

shawnhsiao97, 2022-04-22:
> My kernel image size is 12 MB, it works well. Would you upload your .config?

my .config is the same with nuvoton_nuc980_iot_defconfig

but I had addition some libraries, tools and others, there will increase my rootfs size.

ychuang3, 2022-04-22:
You can move rootfs from ramfs to other stoage such as NAND.

shawnhsiao97, 2022-04-25:
> You can move rootfs from ramfs to other stoage such as NAND.

I would like use DDR if I can, I'm confused why I can only use so little space size?