OpenNuvoton/NUC980-linux-4.4.y #95: 请教关于usb_wwan.c驱动文件，不同内核版本的这个文件为啥不一样？并且新版有问题。

redworlf007, 2022-04-20:
位于这个位置nuc980bsp/linux-4.4.x/drivers/usb/serial的usb_wwan.c文件

4.4.179内核版本里面：
[image]

4.4.289内核版本里面：
[image]

我在移植4G模块驱动，厂家的驱动移植说明上：
[image]

如果按照4.4.289内核版本里面的程序，编译不过。

我想请教下，4.4.289为啥改成这个样子？

我把4.4.289内核里面的文件改回4.4.179，编译才能通过。

ychuang3, 2022-04-20:
我試著 compile ，在 4.4.289 是可以成功編譯的

redworlf007, 2022-04-21:
[Quectel_EC200S_Linux_USB_Driver_User_Guide_V1.0.pdf](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8526777/Quectel_EC200S_Linux_USB_Driver_User_Guide_V1.0.pdf)

> 我試著 compile ，在 4.4.289 是可以成功編譯的

你看看这个文档，我是179上改了编译成功，289上编译不过。

ychuang3, 2022-04-21:
舊版 driver 寫法在較新版本編譯出問題，這很正常，Linux kernel一直持續在做修改。
Quectel 的 work-around 應該是在較舊版本上面驗證的，如果在新版本編譯不過，應該去詢問 Quectel 才對。
此 issue 其實與 NUC980 Linux 4.4 完全無關，建議您去 Quectel 討論區詢問

redworlf007, 2022-04-21:
> 舊版 driver 寫法在較新版本編譯出問題，這很正常，Linux kernel一直持續在做修改。 Quectel 的 work-around 應該是在較舊版本上面驗證的，如果在新版本編譯不過，應該去詢問 Quectel 才對。 此 issue 其實與 NUC980 Linux 4.4 完全無關，建議您去 Quectel 討論區詢問

行吧，我把289里面的相关文件改成了179里面的，然后编译过了，暂时先用着吧，我再去问问Quectel ，谢谢你的耐心解答。