OpenNuvoton/NUC980-linux-4.4.y #96: 请教NUC980DF71yc的系统显示DDR大小的问题。

redworlf007, 2022-04-21:
1、NUC980DF71YC是128M的ddr。
2、系统启动的时候uboot打样是128MDDR
3、但是Linux里面free和top查的都是64Mddr
[image]
[image]
请教一下是啥问题？谢谢。

ychuang3, 2022-04-21:
如果有開 device tree，請在對應 board，例如  arch/arm/boot/dts/nuc980-iot-v1.0.dts
找到
chosen {
		bootargs = "root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M lpj=744448";
	};
將 64M 改為 128M

如果沒開 device tree，請到 menuconfig 的 boot options 裡面去修改成 128M

redworlf007, 2022-04-21:
> 

 

> 如果有開 device tree，請在對應 board，例如 arch/arm/boot/dts/nuc980-iot-v1.0.dts 找到 chosen { bootargs = "root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M lpj=744448"; }; 將 64M 改為 128M
> 
> 如果沒開 device tree，請到 menuconfig 的 boot options 裡面去修改成 128M

我已经这么改了，我想也是这里的问题，谢谢了。