OpenNuvoton/NUC980-linux-4.4.y #97: QSPI Flash问题：程序运行一段时间后，运行sqlite3应用程序无法对Flash读写数据

QHxiedong, 2022-04-21:
我用的是Nor flash W25Q256，mount挂载到文件系统后，sqlite3应用程序运行一段时间后感觉Flash出现问题了，无法读写flash数据，用ls命令卡死，无法看到路径下的文件。
内核版本已经更新到最新的4.4.289了。
启动打印信息如下：
[image]

flash卡死：
[image]

QHxiedong, 2022-04-22:
我在内核配置里面去掉了PDMA功能，问题不出现了。
内核配置如下：
[image]

**下面是带PDMA和不带PDMA时的dmesg信息对比截图**
有PDMA时：
[image]

去掉PDMA后：
[image]

QHxiedong, 2022-04-25:
能否帮忙看下这个问题，谢谢！产品要量产了，压力比较大。

ychuang3, 2022-04-25:
我們沒有你的測試環境，無法復現問題
是不是可以在 driver 加一些 debug 訊息，先看看 driver 運作狀況
或者找當地新唐FAE support

M16AK47MMM, 2022-05-24:
- [ ]请问楼主后面有解决这个Flash读写问题吗？我调试NUC980，对其进行频繁读写文件，也出现了系统挂死。

QHxiedong, 2022-05-24:
我没有继续定位这个问题了，我目前是这样做的，暂时没有问题：
1. 更新到最新内核版本
2. 内核配置QSPI No PDMA

M16AK47MMM, 2022-05-24:
最新版4.4.289内核源代码无法编译，需要做什么配置吗？