OpenNuvoton/NUC980-linux-4.4.y #98: error: ‘CONFIG_PAGE_OFFSETUL’ undeclared (first use in this function)

mx568549060, 2022-04-22:
我用NUC980-linux-4.4.y的内核源码来编写一个驱动程序，遇到报CONFIG_PAGE_OFFSETUL CONFIG_PHYS_OFFSETUL 之类的UL的问题，是我的Makefile哪里没写正确吗？

```
In file included from /home/mx/voip/NUC980-linux-4.4.y/include/linux/list.h:7:0,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/module.h:9,
                 from arch/ztap/kernel/vp_firmware.c:5:
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h: In function ‘__virt_to_phys’:
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:32:25: error: ‘CONFIG_PAGE_OFFSETUL’ undeclared (first use in this function)
 #define PAGE_OFFSET  UL(CONFIG_PAGE_OFFSET)
                         ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:32:22: note: in expansion of macro ‘UL’
 #define PAGE_OFFSET  UL(CONFIG_PAGE_OFFSET)
                      ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:237:26: note: in expansion of macro ‘PAGE_OFFSET’
  return (phys_addr_t)x - PAGE_OFFSET + PHYS_OFFSET;
                          ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:32:25: note: each undeclared identifier is reported only once for each function it appears in
 #define PAGE_OFFSET  UL(CONFIG_PAGE_OFFSET)
                         ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:32:22: note: in expansion of macro ‘UL’
 #define PAGE_OFFSET  UL(CONFIG_PAGE_OFFSET)
                      ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:237:26: note: in expansion of macro ‘PAGE_OFFSET’
  return (phys_addr_t)x - PAGE_OFFSET + PHYS_OFFSET;
                          ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:135:29: error: ‘CONFIG_PHYS_OFFSETUL’ undeclared (first use in this function)
 #define PLAT_PHYS_OFFSET UL(CONFIG_PHYS_OFFSET)
                             ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:135:26: note: in expansion of macro ‘UL’
 #define PLAT_PHYS_OFFSET UL(CONFIG_PHYS_OFFSET)
                          ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:232:21: note: in expansion of macro ‘PLAT_PHYS_OFFSET’
 #define PHYS_OFFSET PLAT_PHYS_OFFSET
                     ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:237:40: note: in expansion of macro ‘PHYS_OFFSET’
  return (phys_addr_t)x - PAGE_OFFSET + PHYS_OFFSET;
                                        ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h: In function ‘__phys_to_virt’:
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:135:29: error: ‘CONFIG_PHYS_OFFSETUL’ undeclared (first use in this function)
 #define PLAT_PHYS_OFFSET UL(CONFIG_PHYS_OFFSET)
                             ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:135:26: note: in expansion of macro ‘UL’
 #define PLAT_PHYS_OFFSET UL(CONFIG_PHYS_OFFSET)
                          ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:232:21: note: in expansion of macro ‘PLAT_PHYS_OFFSET’
 #define PHYS_OFFSET PLAT_PHYS_OFFSET
                     ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:242:13: note: in expansion of macro ‘PHYS_OFFSET’
  return x - PHYS_OFFSET + PAGE_OFFSET;
             ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:32:25: error: ‘CONFIG_PAGE_OFFSETUL’ undeclared (first use in this function)
 #define PAGE_OFFSET  UL(CONFIG_PAGE_OFFSET)
                         ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:32:22: note: in expansion of macro ‘UL’
 #define PAGE_OFFSET  UL(CONFIG_PAGE_OFFSET)
                      ^
/home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/memory.h:242:27: note: in expansion of macro ‘PAGE_OFFSET’
  return x - PHYS_OFFSET + PAGE_OFFSET;
                           ^
In file included from /home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/page.h:163:0,
                 from /home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/asm/thread_info.h:17,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/thread_info.h:58,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/asm-generic/preempt.h:4,
                 from /home/mx/voip/NUC980-linux-4.4.y/arch/arm/include/generated/asm/preempt.h:1,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/preempt.h:59,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/spinlock.h:50,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/seqlock.h:35,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/time.h:5,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/stat.h:18,
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/module.h:10,
[... 99 lines cut ...]
                 from /home/mx/voip/NUC980-linux-4.4.y/include/linux/module.h:24,
                 from arch/ztap/kernel/vp_firmware.c:5:
/home/mx/voip/NUC980-linux-4.4.y/include/asm-generic/module.h:9:8: note: originally defined here
 struct mod_arch_specific
        ^
In file included from arch/ztap/kernel/vp_firmware.c:8:0:
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h:251:42: error: ‘KMALLOC_SHIFT_HIGH’ undeclared here (not in a function)
 extern struct kmem_cache *kmalloc_caches[KMALLOC_SHIFT_HIGH + 1];
                                          ^
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h: In function ‘kmalloc_index’:
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h:236:32: error: ‘KMALLOC_SHIFT_LOW’ undeclared (first use in this function)
 #define KMALLOC_MIN_SIZE (1 << KMALLOC_SHIFT_LOW)
                                ^
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h:269:14: note: in expansion of macro ‘KMALLOC_MIN_SIZE’
  if (size <= KMALLOC_MIN_SIZE)
              ^
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h: In function ‘kmalloc_size’:
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h:236:32: error: ‘KMALLOC_SHIFT_LOW’ undeclared (first use in this function)
 #define KMALLOC_MIN_SIZE (1 << KMALLOC_SHIFT_LOW)
                                ^
/home/mx/voip/NUC980-linux-4.4.y/include/linux/slab.h:477:16: note: in expansion of macro ‘KMALLOC_MIN_SIZE’
  if (n == 1 && KMALLOC_MIN_SIZE <= 32)
                ^
make: *** [Makefile:151: obj/arch/ztap/kernel/vp_firmware.o] Error 1
```

ychuang3, 2022-04-24:
請問你寫的是哪個 driver?
這個嗎  arch/ztap/kernel/vp_firmware.c  ?  kernel 沒有 arch/ztap 這個目錄
而且 arch 應該是 arm 才對

mx568549060, 2022-04-24:
> 請問你寫的是哪個 driver? 這個嗎 arch/ztap/kernel/vp_firmware.c ? kernel 沒有 arch/ztap 這個目錄 而且 arch 應該是 arm 才對

你好，这是我自己写的一个slic的驱动，arch/ztap/kernel/vp_firmware.c是该驱动的代码，这个是没问题的

ychuang3, 2022-04-25:
arch 目錄下是放 CPU architecture 的，ztap 不會是一個 CPU architecture
一個驅動，應該是放置到 driver 目錄下
你應該看看同樣性質的驅動，其他vendor是怎麼做的，放在哪個驅動目錄下
如果你的裝置與驅動是獨一無二的，可以放到 drivers/misc 目錄下，同樣需要看看其他 vendor 是怎麼做了