OpenNuvoton/NUC980-linux-4.4.y #100: 如何设置内核从SPI NAND启动

M16AK47MMM, 2022-05-26:
按照官方Linux-4.4_BSP设置SPI NAND，但每次上电依然检测SPI norFlash，没有识别SPI NAND W25N01GV型号，请问在配置内核时有哪些细节需要注意？
[image]

cwweng-nuvoton, 2022-05-26:
請打開 Device Drivers -> Staging drivers -> SPINAND Device Support
[image]

M16AK47MMM, 2022-05-26:
Staging drivers配置我全部打开了的，但依然无法识别SPI NAND W25N01GV型号
[image]
[image]

M16AK47MMM, 2022-05-26:
[image]

cwweng-nuvoton, 2022-05-26:
請提供你的 kernel config

M16AK47MMM, 2022-05-27:
[config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8785051/config.txt)

M16AK47MMM, 2022-05-27:
按照正常的Linux内核图形界面配置，在启动时仍无法识别到SPI NANDFlash，是否需要对源代码进行修改。

cwweng-nuvoton, 2022-05-27:
arch/arm/mach-nuc980/dev.c 當中定義 使用 SPI NOR 或 SPI NAND  驅動
[image]
如果有定義 CONFIG_BOARD_IOT, CONFIG_BOARD_ETH2UART, CONFIG_BOARD_LORA 當中一個, 就會啟動 SPI NAND 驅動

你的 config 都沒有定義 CONFIG_BOARD_IOT, CONFIG_BOARD_ETH2UART, CONFIG_BOARD_LORA  當中的任何一個
[image]
所以就會啟動 SPI NOR 驅動,  而非 SPI NAND  驅動

closed 2022-05-31 by M16AK47MMM