OpenNuvoton/NUC980-linux-4.4.y #101: Unable to Start Linux Kernel (No Error)

RSTurgay, 2022-05-26:
Hi there,

**It won't boot when booting Linux OS via NAND Flash. I am not getting any error codes either. I am encountering this error in this card that I produced myself. I did not encounter a problem with the same image on the NUC980 ETH 2 card. Why do you think it might?**

`NUC980 IBR 20180813
Boot from NAND
DDR-OK
finish NAND dow▒c{
NAND boot!        #S

U-Boot 2016.11 (May 24 2022 - 18:05:59 +0300)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  256 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x200000, size 0x1400000
 20971520 bytes read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.289
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2271312 Bytes = 2.2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starti▒Booting Linux on physical CPU 0x0
Linux version 4.4.289 (turgay@turgay-VirtualBox) (gcc version 4.8.5 (Buildroot 2016.11.1-g7116cea) ) #2 PREEMPT Thu Apr 14 11:42:52 +03 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
debug: ignoring loglevel setting.
Memory policy: Data cache writeback
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat c0488638, node_mem_map c3f77000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16384 pages, LIFO batch:3
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0 rdinit=/sbin/init mem=64M mtdparts=nand0:0x200000@0x0(u-boot),0x1400000@0x200000(kernel),-(user) ignore_loglevel
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 60068K/65536K available (3342K kernel code, 203K rwdata, 924K rodata, 140K init, 200K bss, 5468K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0432d94   (4268 kB)
      .init : 0xc0433000 - 0xc0456000   ( 140 kB)
      .data : 0xc0456000 - 0xc0488e40   ( 204 kB)
       .bss : 0xc0488e40 - 0xc04baed0   ( 201 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
NR_IRQS:545
clocksource: nuc980-timer5: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 62215505635 ns
sched_clock: 24 bits at 120kHz, resolution 8333ns, wraps every 69905062489ns
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x8400 - 0x843c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource nuc980-timer5
NET: Registered protocol family 2
IP idents hash table entries: 2048 (order: 2, 16384 bytes)
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
NetWinder Floating Point Emulator V0.97 (double precision)
yaffs: yaffs Installing.
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
io scheduler noop registered (default)
nuc980_gpio_probe - pdev = nuc980-gpio
nuc980_dma_probe - pdev = nuc980-dma
nuc980-dma nuc980-dma: NUC980 DMA ready
nuc980-uart.0: ttyS0 at I/O 0x0 (irq = 36, base_baud = 750000) is a NUC980
nuc980-uart.1: ttyS1 at I/O 0x0 (irq = 37, base_baud = 9375000) is a NUC980
nuc980-uart.2: ttyS2 at I/O 0x0 (irq = 38, base_baud = 9375000) is a NUC980
nuc980-uart.3: ttyS3 at I/O 0x0 (irq = 43, base_baud = 9375000) is a NUC980
nuc980-uart.4: ttyS4 at I/O 0x0 (irq = 39, base_baud = 9375000) is a NUC980
nuc980-uart.5: ttyS5 at I/O 0x0 (irq = 44, base_baud = 9375000) is a NUC980
nuc980-uart.6: ttyS6 at I/O 0x0 (irq = 40, base_baud = 9375000) is a NUC980
nuc980-uart.7: ttyS7 at I/O 0x0 (irq = 45, base_baud = 9375000) is a NUC980
nuc980-uart.8: ttyS8 at I/O 0x0 (irq = 41, base_baud = 9375000) is a NUC980
brd: module loaded
RX nuc980_spi0_probe: dma0chan0 module removed
TX nuc980_spi0_probe: dma0chan1 module removed
m25p80 spi1.0: unrecognized JEDEC id bytes: ff, ff, ff
m25p80: probe of spi1.0 failed with error -2
libphy: nuc980_rmii0: probed
libphy: nuc980_rmii1: probed
usbcore: registered new interface driver rndis_wlan
usbcore: registered new interface driver zd1201
usbcore: registered new interface driver asix
usbcore: registered new interface driver ax88179_178a
usbcore: registered new interface driver cdc_ether
usbcore: registered new interface driver net1080
usbcore: registered new interface driver rndis_host
usbcore: registered new interface driver cdc_subset
usbcore: registered new interface driver zaurus
usbcore: registered new interface driver cdc_ncm
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
nuc980-ehci nuc980-ehci: Nuvoton NUC980 EHCI Host Controller
nuc980-ehci nuc980-ehci: new USB bus registered, assigned bus number 1
nuc980-ehci nuc980-ehci: irq 23, io mem 0xb0015000`

Here it waits and there is no more.
The same image file works on ETH2 Board.

scottzhang, 2022-05-30:
you can add printk to see where hangs. For the moment, you hangs when usb-controller is starting up.

RSTurgay, 2022-05-30:
Hi,

I figured out where the problem is. Pin 121 on my own card I developed was stuck because 1.2V supply was given instead of 3.3V. Problem is solved.

Thank you for your interest.

closed 2022-05-30 by RSTurgay