OpenNuvoton/NUC980-linux-4.4.y #104: 64bit time 兼容性（2038问题）？

xray-bit, 2022-06-21:
是否有考虑做 64 位时间的兼容性 patch？

xray-bit, 2022-06-23:
您好，
由于近期做了一些开源软件的交叉编译，可能由于 Nuvoton 官方提供的基于 uClibc 的 toolchain 太老，导致交叉编译失败。所以我最近尝试使用 Buildroot (2022.05) 编译了一套基于 musl (1.2.3) 的 toolchain。
使用 musl toolchian 编译的内核，在 `evdev/input` 上出现了不兼容的情况。不兼容的原因是：musl 为了对应 "2038 年"(y2038) 的问题，开始使用 64 位长度的 `time_t` 类型，用户空间使用该内核 API 时，获取的是 64 位 `time_t`，而当前内核空间返回的是 32 位 `time_t`。

[musl 关于使用 64 位 `time_t` 的说明](https://musl.libc.org/time64.html)

> As of Linux 4.19, all important kernel header fixes except those needed by asound/ALSA are upstream.

由于我对内核的工作不太了解，似乎 4.4 之前的内核还没有过渡到 64 位 `time_t`。
考虑到我们基于 NUC980 的产品是属于工控产品，设计寿命较长，担心 32 位 `time_t` 到 2038 年可能会发生灾难性后果（文件系统等），所以想询问一下官方有没有做修复这个问题的计划。

ychuang3, 2022-06-23:
NUC980 有提供 Buildroot ==>    https://github.com/OpenNuvoton/NUC970_Buildroot
#make nuvoton_nuc980_defconfig  
#make

關於 2038 問題，我們正在 porting NUC980 到 Linux 5.10，最快 Q3/E release，而 Linux 5.10 對此問題有 solution。
https://linux.slashdot.org/story/20/10/17/2237236/linux-510-solves-the-year-2038-problem-until-2486

xray-bit, 2022-06-24:
好的，谢谢！

closed 2022-06-24 by xray-bit