OpenNuvoton/NUC980-linux-4.4.y #106: Error after enable device tree support NUC980

hongduc10, 2022-06-28:
Currently, I have the NUC980 ETH2UART board. I want to use the device tree support for the board, So I have enable Flattened device tree support in linux kernel menuconfig, but after boot, the kernel stuck at dma init

**Env**
```
baudrate=115200
bootdelay=3
stderr=serial
stdin=serial
stdout=serial
setspi=sf probe 0 30000000;
loadkernel=sf read 0x7FC0 0x200000 0x800000
loaddtb=sf read 0x1400000 0xB0000 0x10000
bootmcmd=bootm 0x7FC0 - 0x1400000
bootcmd=run setspi;run loadkernel;run loaddtb;run bootmcmd;
bootargs=root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M lpj=744448

```

**Log**
```
U-Boot 2016.11 (Jun 22 2022 - 17:55:29 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  NAND Flash not found !
NUC980 NAND CONTROLLER IS NOT SUPPORT THE PAGE SIZE. (0, 0)
0 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x800000
SF: 8388608 bytes @ 0x200000 Read: OK
device 0 offset 0xb0000, size 0x10000
SF: 65536 bytes @ 0xb0000 Read: OK
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-4.4.289
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    6046120 Bytes = 5.8 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
SF: Device▒Booting Linux on physical CPU 0x0
Linux version 4.4.289 (user@ubuntu) (gcc version 4.8.5 (Buildroot 2016.11.1-g711                                                                                                             6cea) ) #4 PREEMPT Mon Jun 27 20:52:40 +07 2022
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC980
Memory policy: Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem                                                                                                             =64M lpj=744448
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 55964K/65536K available (3806K kernel code, 223K rwdata, 1116K rodata, 3                                                                                                             548K init, 219K bss, 9572K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
[... 102 lines cut ...]
[<c01da4fc>] (bus_for_each_dev) from [<c01dbc60>] (driver_attach+0x20/0x28)
 r6:c08717b0 r5:c30029c0 r4:c086dc6c
[<c01dbc40>] (driver_attach) from [<c01db8e0>] (bus_add_driver+0x188/0x210)
[<c01db758>] (bus_add_driver) from [<c01dc980>] (driver_register+0x80/0x100)
 r7:c04f782c r6:c3bd52c0 r5:c0851a60 r4:c086dc6c
[<c01dc900>] (driver_register) from [<c01dda94>] (__platform_driver_register+0x3                                                                                                             8/0x4c)
 r5:c0851a60 r4:c0851a60
[<c01dda5c>] (__platform_driver_register) from [<c04e87f0>] (nuc980_dma_driver_i                                                                                                             nit+0x18/0x20)
[<c04e87d8>] (nuc980_dma_driver_init) from [<c00095c8>] (do_one_initcall+0x88/0x                                                                                                             1f0)
[<c0009540>] (do_one_initcall) from [<c04d7e3c>] (kernel_init_freeable+0x12c/0x1                                                                                                             f0)
 r10:00000076 r9:c04f7838 r8:c04fdc20 r7:c04f782c r6:c0885ec0 r5:c0885ec0
 r4:00000006
[<c04d7d10>] (kernel_init_freeable) from [<c03b9784>] (kernel_init+0x10/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03b9774
 r4:00000000
[<c03b9774>] (kernel_init) from [<c000fe18>] (ret_from_fork+0x14/0x3c)
 r4:00000000 r3:ffffffff
Code: 1a00000c e5973014 e3130001 1a000009 (e7f001f2)
---[ end trace 6a872d98887cd521 ]---
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
```

[linux-kernel.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8996848/linux-kernel.txt)

cwweng-nuvoton, 2022-06-28:
The message, " can't get pwm id from dt " occurred in pwm driver while getting the value of pwm id that is defined in device tree.
Did you load .dtb file while running Linux kernel ?

hongduc10, 2022-06-28:
I have load nuc980-eth2uart.dtb to 0xB00000.

cwweng-nuvoton, 2022-06-28:
nuc980-eth2uart.dtb should be loaded at 0x1400000

hongduc10, 2022-06-28:
I download to SPI nand: 0xB0000
And load to RAM from 0xB0000  to 0x1400000 

loaddtb=sf read 0x1400000 0xB0000 0x10000
bootmcmd=bootm 0x7FC0 - 0x1400000

cwweng-nuvoton, 2022-06-28:
Please provide your kernel config (.config) , nuc980.dtsi and nuc980-eth2uart.dts

hongduc10, 2022-06-28:
nuc980.dtsi
[nuc980.dtsi.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8998079/nuc980.dtsi.txt)

nuc980-eth2uart.dts
[nuc980-eth2uart.dts.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8998081/nuc980-eth2uart.dts.txt)

Linux kernel config
[config.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/8998089/config.txt)

cwweng-nuvoton, 2022-06-28:
I burn uboot,  Linux kernel, .dtb file into SPI NAND Flash by NuWriter.
Linux kernel is at 0x200000 of SPI NAND and .dtb file is at 0xD00000

Then boot from SPI NAND. Following are the U-Boot commands I used to boot Linux kernel.
[image]

Linux kernel boot successfully. 
Boot log as attched. 
[nuc980_uboot_bootm_dtb.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/9000435/nuc980_uboot_bootm_dtb.txt)

hongduc10, 2022-06-28:
Do I need to modify the Uboot config to enable device tree?

hongduc10, 2022-06-28:
Could you send me the uboot and uboot spl binary

cwweng-nuvoton, 2022-06-28:
I build U-Boot by nuc980_eth2uart_defconfig.
Attatched are U-Boot SPL and U-Boot bin file that I build today.
[U-Boot.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/9000557/U-Boot.zip)

hongduc10, 2022-06-29:
I have tried to:
- build u-boot with nuc980_eth2uart_defconfig --> enable CONFIG_FIT and CONFIG_OF_LIBFDT
- build linux kernel with nuc980_eth2uart_defconfig --> enable Flattened device tree support
- and enable device tree build in build root
--> now the problem solved
Thank @cwweng-nuvoton

closed 2022-06-29 by hongduc10