OpenNuvoton/NUC980-linux-4.4.y #108: Kernel panic during running

hongduc10, 2022-07-19:
I have run an application and sometimes the kernel is panic show as log.

```

bad: scheduling from the idle thread!
CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.207 #3
Hardware name: NUC980
Backtrace:
[<c0012e40>] (dump_backtrace) from [<c001302c>] (show_stack+0x18/0x1c)
 r6:00000001 r5:c0444f28 r4:c043fb10 r3:00000000
[<c0013014>] (show_stack) from [<c017e298>] (dump_stack+0x20/0x28)
[<c017e278>] (dump_stack) from [<c00410e4>] (dequeue_task_idle+0x54/0x80)
[<c0041090>] (dequeue_task_idle) from [<c003cde8>] (deactivate_task+0x64/0x68)
[<c003cd84>] (deactivate_task) from [<c03178d0>] (__schedule+0x1d0/0x470)
 r6:00000000 r5:c0444f28 r4:c043fb10 r3:00000002
[<c0317700>] (__schedule) from [<c0317bc4>] (schedule+0x54/0xb8)
 r8:ffffffff r7:00000002 r6:c043fb10 r5:c04aa0f8 r4:c043c000 r3:00000001
[<c0317b70>] (schedule) from [<c0317f04>] (schedule_preempt_disabled+0x28/0x44)
 r4:c043c000 r3:c043c000
[<c0317edc>] (schedule_preempt_disabled) from [<c0319758>] (__mutex_lock_slowpat                                                                                                                                                                                                                                             h+0xac/0x17c)
[<c03196ac>] (__mutex_lock_slowpath) from [<c0319860>] (mutex_lock+0x38/0x3c)
 r9:c043dde0 r8:c380b970 r7:0322a000 r6:c043ddc8 r5:c3b66600 r4:c38ab574
[<c0319828>] (mutex_lock) from [<c01a9cc4>] (hw_submit+0x234/0x3c0)
[<c01a9a90>] (hw_submit) from [<c01a935c>] (nuc980_dma_tx_submit+0x78/0x7c)
 r6:c043ddc8 r5:00000798 r4:c38ab574 r3:c01a9a90
[<c01a92e4>] (nuc980_dma_tx_submit) from [<c01cac5c>] (nuc980_prepare_RX_dma+0x1                                                                                                                                                                                                                                             4c/0x1b8)
 r5:c04ab3c8 r4:c04ab2a4
[<c01cab10>] (nuc980_prepare_RX_dma) from [<c01cadb0>] (nuc980_Rx_dma_callback+0                                                                                                                                                                                                                                             xe8/0x110)
 r7:0000cde9 r6:0000001b r5:c04ab490 r4:c04ab2a4
[<c01cacc8>] (nuc980_Rx_dma_callback) from [<c01a92bc>] (nuc980_dma_tasklet+0x12                                                                                                                                                                                                                                             8/0x150)
 r8:c04ab490 r7:c01cacc8 r6:00000002 r5:c043de20 r4:c38ab574 r3:c043de20
[<c01a9194>] (nuc980_dma_tasklet) from [<c00213b0>] (tasklet_action+0x7c/0xe4)
 r9:c047e720 r8:c043c000 r7:00000000 r6:c047e700 r5:c04434c0 r4:00000000
[<c0021334>] (tasklet_action) from [<c002164c>] (__do_softirq+0xac/0x224)
 r7:40000006 r6:00000006 r5:c047e738 r4:00000000
[<c00215a0>] (__do_softirq) from [<c0021a88>] (irq_exit+0xa8/0x100)
 r10:c04348c4 r9:c043e000 r8:c047dee1 r7:c043df3c r6:c3abf780 r5:c0468a68
 r4:00000000
[<c00219e0>] (irq_exit) from [<c004baf8>] (__handle_domain_irq+0x64/0xa8)
 r4:00000000 r3:c043c000
[<c004ba94>] (__handle_domain_irq) from [<c0009024>] (asm_do_IRQ+0x24/0x28)
 r6:f0000000 r5:60000013 r4:c0010c94 r3:c043df08
[<c0009000>] (asm_do_IRQ) from [<c0013a30>] (__irq_svc+0x30/0x70)
Exception stack(0xc043df08 to 0xc043df50)
df00:                   00000000 0005317f 0005217f 60000013 c043c000 c043e06c
df20: c043e06c c047dee1 c047dee1 c043e000 c04348c4 c043df64 600000d3 c043df58
df40: c0010c9c c0010c94 60000013 ffffffff
[<c0010c68>] (arch_cpu_idle) from [<c0044da0>] (default_idle_call+0x28/0x3c)
[<c0044d78>] (default_idle_call) from [<c0044e84>] (cpu_startup_entry+0xd0/0x128                                                                                                                                                                                                                                             )
[<c0044db4>] (cpu_startup_entry) from [<c0316814>] (rest_init+0x78/0x90)
 r7:ffffffff r3:c043c000
[<c031679c>] (rest_init) from [<c0419c50>] (start_kernel+0x310/0x370)
 r4:00000000 r3:c043c000
[<c0419940>] (start_kernel) from [<00008048>] (0x8048)
```

ychuang3, 2022-07-19:
The exception occurred from the kernel thread of NUC980 PDMA driver. It use mutex_lock() in IRQ context, which induce re- shcduling and result in exception.

There's two possible solutions:
1. Remove all mutex_lock() and mutext_unlock() in  drivers\dma\nuc980_dma.c.  or
2. In Linux kernel menuconfig, select 
     Kernel Features --> Preemption Model  --> "No Forced Preemption (Server)"

hongduc10, 2022-07-20:
I have tried the solutions: Kernel Features --> Preemption Model --> "No Forced Preemption (Server)"
But the panic come faster.
Brief my app:
 - Run MQTTS client
 - Use serial port (3 port)
 - Create 5 thread (3 thread for serial ports and 2 thread for MQTTS)

ychuang3, 2022-07-20:
The PDMA is called from UART driver. You can disable UART PDMA.

hongduc10, 2022-07-20:
Does the performace reduce when I disable UART PDMA, or any side effect?

hongduc10, 2022-07-20:
After I tried to disable PDMA of UART: 
Before: Avg CPU is about 3%
After: Avg CPU is about 15%.

Do you have any solution to keep PDMA but kernel is not panic?

I have not try **Remove all mutex_lock() and mutext_unlock() in drivers\dma\nuc980_dma.c.**. Is this safe?

ychuang3, 2022-07-20:
The PDAM driver had an update for the similar issue at 11 months ago.
Some other users report the same issue and fixed at that time.
Both you and I are using the out-of-date version.
Please remove your buidroot and clone again to get the up-to-date version, including kernel.

hongduc10, 2022-07-20:
I checked the git Buildroot, it seem that the lastest commit is almost 2 years ago
[image]

hongduc10, 2022-07-20:
It seem that my kernel is outdate, I will update kernel and check.

Thank you

closed 2022-08-02 by hongduc10