OpenNuvoton/NUC980-linux-4.4.y #116: Watchdog Timer ON 後的2個問題

kenlee0823, 2022-11-25:
**目標板: NK-980ETH2P 2020.02.27 V1.2**

將SW1旁側PG3的R24(10K)電阻移除，Watchdog Timer ON
USB連線查看設置參數:
`PowerOn setting 0x7f6001dc`

dc: 1011 1100，bit3(cfg3)=1，啟用確認

### **問題1: 實測在uboot內閒置超時許久也不會重置?**
"NUC980 U-Boot v2016_11 使用手冊" 章節 "13.3 Watchdog timer 的時間長度"

> 當 Watchdog timer 功能打開後,系統閒置超過 14 秒之後, Watchdog timer 會重置系統,U-Boot 重新開機，
> 每當用戶在 U-Boot 命令列下完成一個命令(輸入 Enter 鍵)之後,會重新計數 14 秒.

uboot 設置:
uboot: nuv980_evb.h
```
    #define CONFIG_HW_WATCHDOG
    #define CONFIG_NUC980_WATCHDOG
```

uboot環境變數:
```
    setenv watchdog on 
    saveenv    
```

### **問題2: 在kernel載入後3x秒發生重置?**
Kernel: menuconfig 配置
```
    Device Drivers  --->
        [*] Watchdog Timer Support  ---->
        -*-   WatchDog Timer Driver Core
        [ ]   Disable watchdog shutdown on close
        < >   Software watchdog
        <*>   Nuvoton NUC980 Watchdog Timer
        <*>     NUC980 WDT wake-up support
        < >   Nuvoton NUC980 Window Watchdog Timer
```
上電後訊息如下:
```
U-Boot 2016.11+ (Nov 24 2022 - 18:00:02 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  0 MiB
jedec aa21SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   nuc980 emac
Warning: eth device name has a space!

Warning: nuc980 emac (eth0) using random MAC address - 8e:8a:41:c2:a3:50

Hit any key to stop autoboot:  0

Loading from nand0, offset 0x200000
   Image Name:   NUC980
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2479872 Bytes = 2.4 MiB
   Load Address: 00008000
   Entry Point:  00008000
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   NUC980
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2479872 Bytes = 2.4 MiB
   Load Address: 00008000
   Entry Point:  00008000
[... 122 lines cut ...]
[    9.793566] devtmpfs: mounted
[    9.798199] Freeing unused kernel memory: 144K
[   10.195866] devpts: called with bogus options
rcS start
[   10.894024] yaffs: dev is 32505859 name is "mtdblock3" rw
[   10.899849] yaffs: passed flags "inband-tags"
Mount /dev/mtdblock3 to /ext successfully.
Sat Jan  1 00:00:00 CST 2005
Starting syslogd: OK
Initializing random number generator: OK
Saving random seed: OK
Starting iptables: OK
Starting network ...
[   13.409941] nuc980-emac0 nuc980-emac0: eth0 is OPENED
Starting myinfod: OK
Starting dropbear(sshd): OK
Starting lighttpd: 2005-01-01 00:00:04: (../src/mod_openssl.c.1333) SSL: inactive/expired X509 certificate '/etc/nuc980.pem'
OK
Starting vsftpd: OK

Welcome to NUC980
iMC-10 login:
```
數秒後發生重置
請問如何修正?

ychuang3, 2022-11-28:
關於問題二
Linux kernel watchdog 開啟之後，需要定時餵狗，如果超時，就會 reset。
請參考範例應用
https://github.com/OpenNuvoton/NUC980_Linux_Applications/blob/master/demos/wdt/wdt.c

ychuang3, 2022-11-28:
問題一: 
U-Boot 不會 reset 是正常的. 
原因是 U-Boot 在等待使用者從 command line 輸入命令時, 會不斷地透過 WATCHDOG_RESET() 
調用 NUC980 Watchdog timer driver 的 hw_watchdog_reset() 來 reset WDT counter
 
[image]

在其他等待周邊費時較久的應用, 例如: 對 SPI NAND Flash 下了 erase 或 write 命令後, 
等待 SPI NAND Flash 的 busy bit. 也是會不斷地調用 WATCHDOG_RESET() 避免系統產生 reset.

kenlee0823, 2022-12-02:
根據你提供的方式,編寫了在啟動腳本中執行的餵狗程序,問題已解決,感謝~