OpenNuvoton/NUC980-linux-4.4.y #118: NUC980，QSPI使用W25Q256JVFQ，开启PDMA后写FLASH系统就死机无响应，关闭后读写正常

youzh213, 2023-02-20:
[nuc980_ebi_defconfig.txt](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/10780190/nuc980_ebi_defconfig.txt)
以上是内核配置，涉及的问题选项是
CONFIG_SPI_NUC980_QSPI0_PDMA=y
该选项只要开启，写FLASH就会死机无响应，设置成
CONFIG_SPI_NUC980_QSPI0_NO_PDMA=y
后就没有问题

youzh213, 2023-02-20:
备注：写FLASH系统就死机，这里的写是直接读写mtdblock时的写

cwweng-nuvoton, 2023-02-20:
我抓最新的 Linux 4.4, 打開 QSPI PDMA
Kernel 開機後, mount JFFS2 file system 讀寫 SPI Flash, 也正常
附件是我的 Kernel config、 編譯出來的 image、.dtb 以及 開機和讀寫 SPI Flash 的 log
[nuc980_qspi_pdma.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/10781956/nuc980_qspi_pdma.zip)

youzh213, 2023-02-22:
> 我抓最新的 Linux 4.4, 打開 QSPI PDMA Kernel 開機後, mount JFFS2 file system 讀寫 SPI Flash, 也正常 附件是我的 Kernel config、 編譯出來的 image、.dtb 以及 開機和讀寫 SPI Flash 的 log [nuc980_qspi_pdma.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/10781956/nuc980_qspi_pdma.zip)

我测试下JFFS2读写也是正常的，就是直接操作MTDBLOCK时死机，你可以试着cat一个文件直接重定向至一个BLOCK
例如：
echo abc > /tmp/test.txt
cat /tmp/test.txt > /dev/mtdblock3
类似这样的操作，然后就死掉了

youzh213, 2023-02-27:
我们供应商联系的工程师已经确认存在这个问题了，正在帮我向你们反馈，有结果我再发布到这里

handsomehui, 2023-03-06:
我们也遇到了相同的问题

cwweng-nuvoton, 2023-03-13:
建議透過  dd  將資料寫入 /dev/mtdN
以下為範例:
dd if=mac.bin of=/dev/mtd3 bs=32 count=1
dd if=/dev/mtd3 of=./test bs=32 count=1

youzh213, 2023-03-29:
:( 问题点不在于证明通过dd操作mtdblock就不死机了，而是我们发现开启PDMA后直接操作mtdblock会死机，因为我们需要向mtdblock读写保存数据，这个应该属于NUC980驱动或者硬件的问题

cwweng-nuvoton, 2023-03-30:
解釋一下為什麼 cat /tmp/test.txt > /dev/mtdblock3 這樣的操作会死机
cat /tmp/test.txt > /dev/mtdblock3 這個操作會調用 內核 MTD 驅動 mtdblock.c 當中的 mtdblock_writesect()
mtdblock_writesect() 通過 vmalloc() 配置一塊 buffer, virtual address 為 0xc4930000, 後續將這塊 buffer 往下傳給 QSPI 驅動.
QSPI 驅動通過 內核 API dma_map_single() 將 virtual address 0xc4930000 轉換為 physical address 0x4930000, 
然後啟動 PDMA, 並等待 PDMA 傳輸完成.
但是 physical address 0x4930000 (大約 73 MB) 超過 DDR 64 MB, 因此 PDMA 傳輸時發現這是個錯誤地址, 而產生 abort
所以 QSPI 驅動會等不到 PDMA 完成傳輸, 造成 cat /tmp/test.txt > /dev/mtdblock3 操作就卡住了

有個解法是 QSPI 驅動 另外配置一塊 buffer 給 PDMA 傳輸使用, 然後再 memcpy() 複製到 MTD 層傳下來的 buffer.
但這會影響效能. 
因此才建議採用 通過 dd 將資料寫入 /dev/mtdN

enginlee, 2023-03-30:
蠻好奇為什麼透過dd就可以正常寫入/dev/mtdN，看了busybox的實作也沒什麼特別的，xmalloc()最後也是導向malloc()得到的記憶體區塊吧?
https://elixir.bootlin.com/busybox/latest/source/coreutils/dd.c
事實上我目前基於NUC980做的產品也是malloc()一塊64KB buffer就fopen()寫入/dev/mtdN，目前也沒有任何問題，所以才好奇這個問題，或許只是巧合?

youzh213, 2023-04-04:
看过你的解答，我明白了，这个应该是vmalloc与kmalloc返回的地址类型导致PDMA硬件进行DMA地址操作时无法访问vmalloc返回的虚拟地址导致的死机，非常感谢你的回答，解开了我的疑惑

closed 2023-04-04 by youzh213