OpenNuvoton/NUC980-linux-4.4.y #122: the size of spi buffer?

AnterCreeper, 2023-05-02:
May i ask for the max size (in bytes) of 
the buffer which i can send in one ioctl?

AnterCreeper, 2023-05-02:
under NUC980-linux-4.4

AnterCreeper, 2023-05-02:
some codes like this: transfer in 32bit mode.
```
void spi_transfer(struct spidev* dev, uint32_t* buffer, int32_t length) {
        if (buffer == NULL) {
                printf("FIXME! empty buffer!\n");
                exit(1);
        }

        int processed = 0;
        const int batch = 1024;
        while (processed < length) {
                int size = min(length - processed, batch);
                dev->buf.len = size * 4;
                dev->buf.tx_buf = (unsigned int)&buffer[processed];
                dev->buf.rx_buf = (unsigned int)NULL;
                if (ioctl(dev->fd, SPI_IOC_MESSAGE(1), &dev->buf) < 1) {
                        printf("FIXME! cannot send spi message!\n");
                        printf("errno:%d %s\n", errno, strerror(errno));
                        exit(1);
                }
                processed = processed + size;
        }
        return;
}
```

AnterCreeper, 2023-05-02:
i have figured out that
1.default buffer size is 4096KBytes, you need to modified it by adding environment args: spidev.bufsiz=65536
2.the size that dma can transfer once is limited in 65536 Bytes.

cwweng-nuvoton, 2023-05-04:
SPI requests PDMA channel from Kernel ant then the channel is dedicated for SPI without sharing with other peripheral.

NUC980 linux-4.4 SPI performance is really about 30 Mbps.

AnterCreeper, 2023-05-05:
> SPI requests PDMA channel from Kernel and then the channel is dedicated for SPI without sharing with other peripheral.
> 
> NUC980 linux-4.4 SPI performance is really about 30 Mbps.

Is there any method to improve the performace?

AnterCreeper, 2023-05-05:
Or only must I use QSPI or SDIO?
It's Chili Board, and AFAIK NUC980DR61YC have no such peripherials.

cwweng-nuvoton, 2023-05-05:
> Is there any method to improve the performace?

We've tried our best to improve NUC980 linux-4.4 SPI performance. But performance can only go so far

AnterCreeper, 2023-05-06:
@cwweng-nuvoton Thanks!