OpenNuvoton/NUC980-linux-4.4.y #124: How do resolve the NUC980 Err-DDR error?

blockchain4trust, 2023-06-14:
[image]
[image]

blockchain4trust, 2023-07-19:
This small batch of NUC980 PCB has a total of 50 pieces, and 5 pieces have Err-DDR, with a 10% defective rate.

The welding fault was ruled out, because this is machine SMT production, there can be no 10% error rate.

There is a clue: after the appearance of Err-DDR, if the nRST<-+-> GND NUC980 occasionally works.

What exactly causes an Err-DDR error? Please provide a process for troubleshooting intentionality, thank you.

ychuang3, 2023-07-19:
It could be related to the board voltage or other issues. Please reflect the problem through the purchasing channel from where you acquired the IC.
Here, we can only provide software BSP-related inquiries and cannot address hardware issues.