OpenNuvoton/NUC980-linux-4.4.y #125: Bad mode in prefetch abort handler detected

haimingweili, 2023-08-15:
使用过程如下：
1. 在内核配置界面启用FIQ功能；
2. 按照文档《NUC980 Linux 4.4 BSP User Manual CHT.pdf》添加fiq中断初始化代码；
3. 其中'init_FIQ(0);'放置在内核函数‘nuc980_init_irq’下； 
4. ARM寄存器初始化函数和中断回调在外部模块，SP堆栈大小使用的是例程中的1024*1；
5. 修改使用定时器的中断优先级为0;
6. 初始化定时器寄存器相关参数使其定时产生中断, 并编译成测试ko模块；
7. 系统完成启动，并手动加载测试模块，出现中断异常；
[image]

ychuang3, 2023-08-15:
做成 .ko 可能有未知問題，請直接加入 kernel
直接修改   arch/arm/mach-nuc980/irq.c   ==> 修改後代碼如下
[irq.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/12342847/irq.zip)

剛剛測試了一下, kernel 可以正常帶起

haimingweili, 2023-08-15:
@ychuang3 根据您提供的示例代码，并且打开设备树后，定时器可以在FIQ模式下正常工作，中断响应优于IRQ，但是对于特殊需求的中断回调，每次更改功能都需要重新烧写内核，较为不便，于是在内核做修改，其中修改方式不规范，修改 arch/arm/kernel/traps.c ==>
extern void (*trap_handle_fiq)(void);
asmlinkage void __exception_irq_entry handle_fiq_as_nmi(struct pt_regs *regs)
{
	// struct pt_regs *old_regs = set_irq_regs(regs);

	// nmi_enter();

	/* nop. FIQ handlers for special arch/arm features can be added here. */
	if (trap_handle_fiq){
		trap_handle_fiq();
	}

	// nmi_exit();

	// set_irq_regs(old_regs);
}
其中屏蔽内核相关调用，然后在回调函数处理中断号，及通知CPU中断结束，内核起来后，加载外部驱动可以工作，但是稳定性需要进一步测试，我还想请问一下，NUC980的IRQ堆栈大小如何设置？

ychuang3, 2023-08-16:
關於 IRQ stack,  NUC980 沒有做任何修改, 使用 Linux kernel arm 的設定
底下這篇文章有一些介紹可以參考
https://blog.csdn.net/weixin_42135087/article/details/107931597

建議不要對 kernel做一些不確定性的修改，後續開發過程如果 kernel crash, 會非常難以 debug

開發過程可以用 NuWriter 直接 download 到 DRAM 執行，不需要做燒寫

haimingweili, 2023-08-16:
@ychuang3 感谢帮助，不确定的kernel异常是致命的，为了稳定性开发，我们将按照建议把FIQ放置在内核使用

closed 2023-08-16 by haimingweili