OpenNuvoton/NUC980-linux-4.4.y #126: spi 的片选都使用怎么配置

zwjin1210, 2023-08-22:
如何配置同时开启spidev1.0 spidev1.1 spidev2.0 spidev2.1 为啥我
[image]
[image]

cwweng-nuvoton, 2023-08-22:
Hi,
請修改 arch/arm/mach-nuc980/dev.c   將 SPI0 和 SPI1 的 SS0 和 SS1 都定義為 "spidev", 
SPI0 bus_num 定義為 1,  SS0 和 SS1 的 chip_select 分別定義為 0 和 1
SPI1 bus_num 定義為 2,  SS0 和 SS1 的 chip_select 分別定義為 0 和 1 
如下
[image]

附上我的 dev.c
[dev.zip](https://github.com/OpenNuvoton/NUC980-linux-4.4.y/files/12404503/dev.zip)

開機後, /dev/ 目錄下即可產生 spidev1.0, spidev1.1, spidev2.0, 和 spidev2.1
[image]

zwjin1210, 2023-08-22:
是的 解决了