OpenNuvoton/NUC980-linux-4.4.y #127: 求助，SD卡挂载出现一些问题

CodeMzc, 2023-08-22:
### 问题描述：

SD卡无论是随内核启动挂载，还是文件系统加载完后再插入，日志都会出现报错，但是最终能够挂载上。
而且SD卡挂载时间长，有2秒左右，公司项目需要缩短Linux系统启动时间。
尝试了以下方法，皆失败：
1、自行修改nuc980_sd.c，打开ENTRY()、LEAVE()这两个宏想调试一下，发现只要打开这两个，拔出SD卡后内核就会挂掉。
2、修改SD卡的width、timing等属性，内核也会直接挂掉，修改频率mmc->f_min直接到25000000HZ，没有明显效果，依然报错。

请问如何解决SD卡挂在报错和加载时间长的问题呢，麻烦各位了，感谢！

### 日志信息：
**启动日志：**
```
//......
[    2.101091] nuc980_sd_probe - pdev = nuc980-sdh
[    2.154708] usbcore: registered new interface driver usbhid
[    2.160241] usbhid: USB HID core driver
[    2.175849] nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
[    2.182441] nand: Winbond W25N01GV 1G 3.3V
[    2.186508] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    2.194383] nand: WARNING: spi0.0: the ECC used on your system is too weak compared to the one required by the NAND chip
[    2.205316] Scanning device for bad blocks
[    2.461341] Creating 3 MTD partitions on "nand0":
[    2.466291] 0x000000000000-0x000000200000 : "u-boot"
[    2.477649] 0x000000200000-0x000001600000 : "Kernel"
[    2.489266] 0x000001600000-0x000008000000 : "user"
[    2.505016] NET: Registered protocol family 17
[    2.516883] nuc980-rtc nuc980-rtc: setting system clock to 2023-08-22 05:31:02 UTC (1692682262)
[    2.555266] yaffs: dev is 32505858 name is "mtdblock2" rw
[    2.560624] yaffs: passed flags "inband-tags"
[    2.601833] mmc0: mmc_rescan_try_freq: trying to init card at 400000 Hz
[    2.672016] VFS: Mounted root (yaffs2 filesystem) on device 31:2.
[    2.680283] devtmpfs: mounted
[    2.685191] Freeing unused kernel memory: 148K
[    3.643958] mmc0: new high speed SDHC card at address 1234
[    3.664933] mmcblk0: mmc0:1234 SA16G 14.4 GiB
[    3.677316]  mmcblk0: p1
Password for 'root' changed
[    6.557866] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
ln: /mnt/mmcblk0p1/mmcblk0p1: Operation not permitted
[Mount VFAT]: /dev/mmcblk0p1 --> /mnt/mmcblk0p1
mount: mounting /dev/mtdblock2 on /mnt/mtdblock2 failed: No such device
[Mount YAFFS2]: /dev/mtdblock2 --> /mnt/mtdblock2
[    7.038183] nuc980-emac0 nuc980-emac0: eth0 is OPENED
```
**mmc日志：**

```
~ # dmesg | grep mmc*
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000] Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
[    0.000716] Console: colour dummy device 80x30
[    2.108891] mmc0: clock 0Hz busmode 2 powermode 1 cs 0 Vdd 21 width 1 timing 0
[    2.122016] mmc0: clock 400000Hz busmode 2 powermode 2 cs 0 Vdd 21 width 1 timing 0
[    2.601833] mmc0: mmc_rescan_try_freq: trying to init card at 400000 Hz
[    2.608424] mmc0: starting CMD52 arg 00000c00 flags 00000195
[    2.774991] mmc0: req done (CMD52): -110: 00000000 00000000 00000000 00000000
[    2.775066] mmc0: starting CMD52 arg 80000c08 flags 00000195
[    2.939166] mmc0: req done (CMD52): -110: 00000000 00000000 00000000 00000000
[    2.939249] mmc0: clock 400000Hz busmode 2 powermode 2 cs 1 Vdd 21 width 1 timing 0
[    2.940508] mmc0: starting CMD0 arg 00000000 flags 000000c0
[    2.940716] mmc0: req done (CMD0): 0: 00000000 00000000 00000000 00000000
[    2.945549] mmc0: clock 400000Hz busmode 2 powermode 2 cs 0 Vdd 21 width 1 timing 0
[    2.946833] mmc0: starting CMD8 arg 000001aa flags 000002f5
[    2.947191] mmc0: req done (CMD8): 0: 000001aa 00000000 00000000 00000000
[    2.947241] mmc0: starting CMD5 arg 00000000 flags 000002e1
[    3.111324] mmc0: req failed (CMD5): -110, retrying...
[    3.275433] mmc0: req failed (CMD5): -110, retrying...
[    3.439541] mmc0: req failed (CMD5): -110, retrying...
[    3.607108] mmc0: req done (CMD5): -110: 00000000 00000000 00000000 00000000
[... 108 lines cut ...]
[    6.554266] mmc0:     (CMD12): 0: 00000b00 00000000 00000000 00000000
[    6.554316] mmc0: req done (CMD18): 0: 00000900 00000000 00000000 00000000
[    6.554341] mmc0:     2560 bytes transferred: 0
[    6.554383] mmc0:     (CMD12): 0: 00000b00 00000000 00000000 00000000
[    6.554541] mmc0: starting CMD18 arg 0000480d flags 000000b5
[    6.554591] mmc0:     blksz 512 blocks 4 flags 00000200 tsac 100 ms nsac 0
[    6.554624] mmc0:     CMD12 arg 00000000 flags 00000095
[    6.555608] mmc0: req done (CMD18): 0: 00000900 00000000 00000000 00000000
[    6.555658] mmc0:     2048 bytes transferred: 0
[    6.555699] mmc0:     (CMD12): 0: 00000b00 00000000 00000000 00000000
[    6.555749] mmc0: req done (CMD18): 0: 00000900 00000000 00000000 00000000
[    6.555783] mmc0:     2048 bytes transferred: 0
[    6.555816] mmc0:     (CMD12): 0: 00000b00 00000000 00000000 00000000
[    6.556016] mmc0: starting CMD18 arg 00004811 flags 000000b5
[    6.556083] mmc0:     blksz 512 blocks 15 flags 00000200 tsac 100 ms nsac 0
[    6.556116] mmc0:     CMD12 arg 00000000 flags 00000095
[    6.557133] mmc0: req done (CMD18): 0: 00000900 00000000 00000000 00000000
[    6.557183] mmc0:     7680 bytes transferred: 0
[    6.557224] mmc0:     (CMD12): 0: 00000b00 00000000 00000000 00000000
[    6.557266] mmc0: req done (CMD18): 0: 00000900 00000000 00000000 00000000
[    6.557299] mmc0:     7680 bytes transferred: 0
[    6.557333] mmc0:     (CMD12): 0: 00000b00 00000000 00000000 00000000
[    6.557866] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

```

ychuang3, 2023-08-23:
SD log 是正常訊息，kernel 會使用不同 command 嘗試判定 SD card type,
這些 command 嘗試會有失敗的, 很正常
SD 加載的時間可能無法優化, 或許試看看不同廠牌的 SD card,

如果要縮短開機時間, 去除 kernel 打印訊息會有幫助,
UART 打印大量訊息會花費不少時間
從 menuconfig "kernel hacking" 進去修改 debug level

CodeMzc, 2023-08-24:
@ychuang3 好的，感谢指点

closed 2023-08-24 by CodeMzc