OpenNuvoton/NUC980-linux-4.4.y #131: spi nand flash怎样读取unique id

xiayudeshijie, 2023-10-22:
使用的是W25N01GVZEIG NAND Flash，看了手册有Unique id，需要读出来用于加密，看了内核驱动，不知道怎样做

cwweng-nuvoton, 2023-10-23:
Hi,
在 drivers/staging/mt29f_spinand/mt29f_spinand.c 的 spinand_read_id() 裡面會讀取 SPI NAND ID
可以在當中打印出 SPI NAND ID
[image]

xiayudeshijie, 2023-10-23:
static int spinand_read_id(struct spi_device *spi_nand, u8 *id)
{
	int retval;
	struct spinand_cmd cmd = {0};

	cmd.cmd = CMD_READ_ID;
	cmd.n_rx = 3;
	cmd.rx_buf = &nand_id[0];

	retval = spinand_cmd(spi_nand, &cmd);
	if (retval < 0) {
		dev_err(&spi_nand->dev, "error %d reading id\n", retval);
		return retval;
	}
	id[0] = nand_id[1];
	id[1] = nand_id[2];
	spinand_parse_id(spi_nand, nand_id, id);
	return retval;
}

这个官网下载的内核的代码，请问你发的这个代码是更改后的吗？    比如说cmd.n_rx = 3; 改为 cmd.n_rx = 4;

cwweng-nuvoton, 2023-10-23:
cmd.n_rx = 3; 改为 cmd.n_rx = 4;
以及新增 id[2] 是在 2023/6/16 的 5bf110e981132 這一版所做的修改
[image]

xiayudeshijie, 2023-10-23:
我测试了，这个代码读的是厂家ID和设备型号的ID，也就是Manufacturer ID: 0xef, Chip ID: 0xaa，这个型号的不同的flash我试了都是一样的，我看了W25N01GVZEIG NAND Flash的手册，里面是有唯一unique id的，这个怎么读呢

cwweng-nuvoton, 2023-10-23:
W25N01GVZEIG SPI NAND Flash的手册上說 Page 0 是 Unique ID Page, 長度是 32 byte * 16
[image]

建議可以透過 U-Boot 的 sf 命令讀取 page 0
下面範例是讀取 page 0 長度 512(0x200) byte 資料到 DDR 0x100000 地址, 然後再透過 md 命令 dump DDR 0x100000 內容
U-Boot> sf probe
U-Boot> sf read 0x100000 0 0x200
U-Boot> md 0x100000

xiayudeshijie, 2023-10-23:
> W25N01GVZEIG SPI NAND Flash的手册上說 Page 0 是 Unique ID Page, 長度是 32 byte * 16 [image]
> 
> 建議可以透過 U-Boot 的 sf 命令讀取 page 0 下面範例是讀取 page 0 長度 512(0x200) byte 資料到 DDR 0x100000 地址, 然後再透過 md 命令 dump DDR 0x100000 內容 U-Boot> sf probe U-Boot> sf read 0x100000 0 0x200 U-Boot> md 0x100000

谢谢，如果只能这样话，那只能想别的办法生成一个唯一的ID作为板子的识别了