OpenNuvoton/NUC980-linux-4.4.y #134: jffs2 Newly-erased block contained word问题

QHxiedong, 2023-12-21:
产品Flash是SPI NOR Flash W25Q256，出现了jffs2 Newly-erased block contained word，如下图所示，请帮忙指导如何查找和解决问题，谢谢！

[image]

cwweng-nuvoton, 2023-12-21:
請問是在 W25Q256 SPI Flash 上 mount JFFS2 root file system 時出現這個錯誤嗎 ?
還是做什麼操作出現的 ?

QHxiedong, 2023-12-21:
mount时还没有出现，在写文件时会出现，读文件时没有出现。

例如下图拷贝log.txt到log.txt1会出现
[image]
但是单板复位再启动，log.txt1并没有被保存下来。如下图：
[image]

cwweng-nuvoton, 2023-12-21:
我仿造你做同樣的拷貝文件,  但是沒有出現你說的錯誤
[image]
板子復位後, log.txt1 也有保存下來

你的內核是最新的嗎 ?  
有沒有做什麼修改 ?

QHxiedong, 2023-12-21:
内核版本是：4.4.289
[image]

不是每台设备都是这样的，生产了900台设备，目前发现只有2台设备出现了这样的问题，剩余的设备都暂时正常工作中。

QHxiedong, 2023-12-21:
最早的设备是22年4月份开始上线运行的

cwweng-nuvoton, 2023-12-22:
請檢查一下 W25Q256 的 Status Register-1 的 Block protect 是否被設起來了
[image]
如果 Block protect 設起來,  erase/write 會出現錯誤

QHxiedong, 2023-12-22:
我代码里添加了这行打印：
[image]
运行打印都是0x0，如下：
[image]

我添加的代码是否存在错误？

QHxiedong, 2023-12-22:
我增加了读取SPINOR_OP_RDCR这个寄存器，貌似Status Register-2的CMP位被置位成1了，整个芯片被protect了？

[image]
[image]
[image]
[image]

cwweng-nuvoton, 2023-12-22:
對.  整個 W25Q256 都被 erase/write protect 了

QHxiedong, 2023-12-22:
1. 导致整个W25Q256被 erase/write protect原因可能是什么呢？
2. 我应该如何解除erase/write protect，能否在应用层解决？

谢谢！

cwweng-nuvoton, 2023-12-22:
同樣在 read_sr() 裡面,  用  nor->write_reg() 將 Status Register-2 的 CMP bit 清為 0
試試看

QHxiedong, 2023-12-22:
通过清除CMP，可以恢复正常了，设置sqlite数据库数据和新建文件夹均正常，重启后文件均保存了。
重启前：
[image]
重启后：
[image]

请问能否在应用层更改CMP的值，可以的话，请帮忙提供下方法。
如果只能在驱动层更改，针对这个问题，最合理的解决方案是什么？
谢谢！

cwweng-nuvoton, 2023-12-24:
在應用層運行 spidev_test,  通過參數 -p 傳送 0x31 command 將 W25Q256 的 status-2 寄存器的 CMP bit 清為 0
[image]

另外, 要記得修改 arch/arm/mach-nuc980/dev.c, 將 QSPI0 原本的  m25p80 註解掉, 改為
[image]
就可以把 W25Q256 當成一般的 SPIDEV 存取

QHxiedong, 2023-12-25:
这个方法也要改内核代码，考虑到设备也都发货到客户手中了，还是选择从驱动代码里面修改。

我写了个CMP位检查并清零的函数，在spi_nor_scan函数中调用，如下所示：

函数：
[image]

调用：
[image]

经过验证是可以的，不知道是否合理，请帮忙指导下，谢谢！

cwweng-nuvoton, 2023-12-25:
客戶會有 erase/write protect 的需求嗎 ?
如果沒有, 這樣的修改 OK.
這樣修改後, 每次啟動都會將 W25Q256 status-2 寄存器的 CMP bit 清為 0

QHxiedong, 2023-12-25:
是的，每次启动都会将CMP清零。客户没有erase/write保护的需求。
现在的方式需要等到重启设备才能解决，请问有更好的方法吗，出现问题时能实时解决掉。
通过dmesg打印的“Node totlen on flash”可以定位到jffs2_mark_node_obsolete函数， “Newly-erased block contained word”定位到jffs2_block_check_erase函数。但是我不知道应该调用什么函数可以将W25Q256的SR_2寄存器的CMP位清零。

closed 2024-01-19 by QHxiedong