OpenNuvoton/NUC980-linux-4.4.y #137: About NUC980 Framebuffer Support using TFT LCD over SPI Interface

RSTurgay, 2024-01-31:
Hi everyone and dear developers,

I want to create GUI application in NUC980 using TFT LCD over SPI. I enabled that Linux Device Drivers Framebuffer support. I see that `/sys/class/graphics/fbcon`. 

I want to create `/dev/fb0` device. It is shuould be TFT LCD like ILI9341 2.8 TFT LCD. 

`https://github.com/notro/fbtft`

It is support ILI9341 chip drivers.

What path should I follow? Somebody help me please. I didn't found any resource this topic.

ychuang3, 2024-02-01:
Please refer to chapter 5.2 of this document:
https://www.nuvoton.com/export/resource-files/zh-tw--UM_NUC980_Linux_environment_on_VMware_EN_Rev1.02.pdf

Note that this is supported in NUC980 Linux-5.10 only and not in NUC980 Linux-4.4.

RSTurgay, 2024-02-01:
> Please refer to chapter 5.2 of this document: https://www.nuvoton.com/export/resource-files/zh-tw--UM_NUC980_Linux_environment_on_VMware_EN_Rev1.02.pdf
> 
> Note that this is supported in NUC980 Linux-5.10 only and not in NUC980 Linux-4.4.

Thank you ychuang3.

closed 2024-02-01 by RSTurgay