OpenNuvoton/NUC980-linux-4.4.y #138: NUC980的SPI驱动问题，导致内核函数死循环

CodeMzc, 2024-06-04:
你好，我们这边在生产的时候突发了一个问题，程序运行一段时间后，CPU异常偏高，结束进程后程序变成僵尸进程，CPU占用依旧很高。
之前一直没有动过SPI的驱动代码，也没出现过这个问题，这次突然发现程序异常，复现也挺困难的。

debug后发现，是linux内核中SPI驱动中的一个while循环导致的，功能是wait busy。
读了一下SPI Status Register这个寄存器发现，REG_STATUS的值是 0x8059e03，Busy Status这一位是1，也就是in busy state，
&1之后等于1，所以才导致死循环。同时RX的FIFO寄存器也显示RX BUFF满了。

代码位置如下：drivers/spi/spi-nuc980-spi0.c 
第358行：
`while(__raw_readl(hw->regs + REG_STATUS) & 1); //wait busy`

把这一行代码修改一下，做了一个超时处理，目前稳定一些，还没有再次出现死循环的问题，还需要更长的时间进行测试。
修改的内容：

```
static int nuc980_spi0_txrx(struct spi_device *spi, struct spi_transfer *t)
{
	struct nuc980_spi *hw = (struct nuc980_spi *)to_hw(spi);
	unsigned int tmp_reg_value = 0;//新增修改
	int retry = 0;//新增修改
#if defined(CONFIG_SPI_NUC980_SPI0_PDMA)
	struct nuc980_ip_dma *pdma=&dma;
	struct nuc980_dma_config dma_crx,dma_ctx;
	dma_cookie_t            cookie;
#elif defined(CONFIG_SPI_NUC980_SPI0_NO_PDMA)

……

	wait_event_interruptible(spi0_slave_done, (spi0_slave_done_state != 0));
	spi0_slave_done_state=0;

	// while(__raw_readl(hw->regs + REG_STATUS) & 1); //wait busy
	/*这是我添加的超时重试处理*/
	while(1)
	{
		tmp_reg_value = __raw_readl(hw->regs + REG_STATUS);
		if(tmp_reg_value & 1)
		{
			retry++;
			if(retry > 100)
			{
				return t->len;
			}
			continue;
		}
		else
		{
			break;
		}
	}

……

｝
```

不知道为什么，SPI正常读写的时候，会出现接收忙的情况。请各位帮忙解答一些，感谢！

CodeMzc, 2024-06-04:
emmm，修改后又SPI是能正常收发了，但系统出现了重启的情况，我继续排查

ychuang3, 2024-06-06:
謝謝您提出的問題，我們已經修改了 spi driver, 加上 timeout 檢查。
請進行更新。

CodeMzc, 2024-06-07:
好的，感谢跟进问题，我尝试一下更新后的代码。

CodeMzc, 2024-06-07:
您好。
我尝试使用更新后的代码，出现了 `spidev spi1.0: SPI transfer failed: -121` ，然后内核宕机，串口调试无法继续输入输出。

在昨天和前天，我使用之前的代码进行debug的时候，也是遇到类似的问题。
重启是因为我们程序中使用了watch dog，所以遇到这个错误系统就复位了。关闭watch dog，系统会宕机，不会重启。
希望您能继续跟进这个问题，非常感谢！

在 drivers/spi/spi-nuc980-spi0.c 第378行的位置。
以下是我添加的debug代码：
```
	end = jiffies + msecs_to_jiffies(SPI_GENERAL_TIMEOUT_MS);
	while (__raw_readl(hw->regs + REG_STATUS) & 1) { //wait busy
		if (time_after(jiffies, end)) {
			printk("SPI wait busy timeout: %d\n", __LINE__);
			printk("%s: REG_CTL value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_CTL));
			printk("%s: REG_CLKDIV value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_CLKDIV));
			printk("%s: REG_PDMACTL value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_PDMACTL));
			printk("%s: REG_FIFOCTL value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_FIFOCTL));
			printk("%s: REG_STATUS value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_STATUS));
			printk("%s: REG_TX value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_TX));
			printk("%s: REG_RX value: 0x%x\n",__func__, __raw_readl(hw->regs + REG_RX));
			return 0;
		}
	}
```

内核日志1：
REG_PDMACTL是0x1

```
[  555.494774] SPI wait busy timeout: 381
[  555.498499] nuc980_spi0_txrx: REG_CTL value: 0x805
[  555.503258] nuc980_spi0_txrx: REG_CLKDIV value: 0x18
[  555.508974] nuc980_spi0_txrx: REG_PDMACTL value: 0x1
[  555.513899] nuc980_spi0_txrx: REG_FIFOCTL value: 0x22000000
[  555.519683] nuc980_spi0_txrx: REG_STATUS value: 0x8059e03
[  555.525224] nuc980_spi0_txrx: REG_TX value: 0x0
[  555.530341] nuc980_spi0_txrx: REG_RX value: 0x82
[  555.535166] spidev spi1.0: SPI transfer failed: -121

```

内核日志2：
REG_PDMACTL是0x3

```
2024-06-07 14:44:48 [core_db/D] exec:select count(*) from dev_S15
2024-06-07 14:44:48 [core_db/D] S15 max store cnt:0,cur cnt:0
2024-06-07 14:44:48 [core_db/D] exec:select count(*) from dev_1
2024-06-07 14:44:48 [core_db/D] 1 max store cnt:0,cur cnt:0
2024-06-07 14:44:48 [core_db/D] exec:select count(*) from soe
[   20.374974] SPI wait busy timeout: 381
[   20.378699] nuc980_spi0_txrx: REG_CTL value: 0x805
[   20.383458] nuc980_spi0_txrx: REG_CLKDIV value: 0x18
[   20.405799] nuc980_spi0_txrx: REG_PDMACTL value: 0x3
[   20.410741] nuc980_spi0_txrx: REG_FIFOCTL value: 0x22000000
[   20.430216] nuc980_spi0_txrx: REG_STATUS value: 0x8059e03
[   20.443191] nuc980_spi0_txrx: REG_TX value: 0x0
[   20.453333] nuc980_spi0_txrx: REG_RX value: 0xb0
[   20.464808] spidev spi1.0: SPI transfer failed: -121

NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
4-bit
finish SPI dow▒load
```

CodeMzc, 2024-06-07:
阅读源码并跟踪代码后发现是来源于drivers/spi/spi.c
此文件中的函数 `static int spi_transfer_one_message(struct spi_master *master,struct spi_message *msg)`
 从第950行开始：

```
static int spi_transfer_one_message(struct spi_master *master,
				    struct spi_message *msg)
{
	struct spi_transfer *xfer;
	bool keep_cs = false;
	int ret = 0;
	unsigned long ms = 1;
	struct spi_statistics *statm = &master->statistics;
	struct spi_statistics *stats = &msg->spi->statistics;

	spi_set_cs(msg->spi, true);

……
                        /* 这一行运行后会返回-121 */
			ret = master->transfer_one(master, msg->spi, xfer);
			if (ret < 0) {
				SPI_STATISTICS_INCREMENT_FIELD(statm,
							       errors);
				SPI_STATISTICS_INCREMENT_FIELD(stats,
							       errors);
				dev_err(&msg->spi->dev,
					"SPI transfer failed: %d\n", ret);
				goto out;
			}

……
	return ret;
}

```
master->transfer_one(master, msg->spi, xfer)这个函数的触发，
是由spi-bitbang.c这个文件的`int spi_bitbang_start(struct spi_bitbang *bitbang)`函数中的
`master->transfer_one = spi_bitbang_transfer_one;` 进行设置

查看spi_bitbang_transfer_one函数：
```
static int spi_bitbang_transfer_one(struct spi_master *master,
				    struct spi_device *spi,
				    struct spi_transfer *transfer)
{
	struct spi_bitbang *bitbang = spi_master_get_devdata(master);
	int status = 0;

	if (bitbang->setup_transfer) {
		status = bitbang->setup_transfer(spi, transfer);
		if (status < 0)
			goto out;
	}

	if (transfer->len)
		status = bitbang->txrx_bufs(spi, transfer);

	if (status == transfer->len)
		status = 0;
	else if (status >= 0)
		status = -EREMOTEIO;

out:
	spi_finalize_current_transfer(master);

	return status;
}
```

在执行 `bitbang->txrx_bufs(spi, transfer);` 这一行时，
txrx_bufs这个函数是在spi-nuc980-spi0.c中由probe函数进行注册：

```
static int nuc980_spi0_probe(struct platform_device *pdev)
{
……
	platform_set_drvdata(pdev, hw);
	init_completion(&hw->done);
	master->mode_bits          = (SPI_MODE_0 | SPI_CS_HIGH | SPI_LSB_FIRST | SPI_CPHA | SPI_CPOL);
	master->dev.of_node        = pdev->dev.of_node;
	master->num_chipselect     = hw->pdata->num_cs;
	master->bus_num            = hw->pdata->bus_num;
	hw->bitbang.master         = hw->master;
	hw->bitbang.setup_transfer = nuc980_spi0_setupxfer;
	hw->bitbang.chipselect     = nuc980_spi0_chipsel;
	hw->bitbang.txrx_bufs      = nuc980_spi0_txrx;
	hw->bitbang.master->setup  = nuc980_spi0_setup;

……
}
```

因为超时处理中return 0，所以导致`status = bitbang->txrx_bufs(spi, transfer);`执行后 status = 0，
```
	if (status == transfer->len)
		status = 0;
	else if (status >= 0)
		status = -EREMOTEIO;

out:
	spi_finalize_current_transfer(master);

	return status;
```
函数最后会返回 -EREMOTEIO，也就是-121，但是我没想明白为什么会引起内核宕机。

cwweng-nuvoton, 2024-06-07:
Hi,

請問要如何復現這個内核宕机的問題 ?

Thanks.

CodeMzc, 2024-06-11:
你好，cwweng

复现这个问题需要依赖硬件，并且时间不确定，我们用SPI连接了一个LoRa模块（LoRa1278），在运行过程中会出现上出问题。
如果不接入这个模块，运行一切正常。

目前是内核崩溃触发看门狗程序，系统重启，初步判断和硬件存在一定的关系。
软件驱动这边有没有方法规避呢，比如说，当出现这个情况的时候，清除某些标志位，让SPI的运行状态变得正常。

希望能得到你的回复，感谢。

cwweng-nuvoton, 2024-06-11:
Hi,

如果有其他方法可以復現問題, 請跟我們說,  我們才能 debug.

根據你提供的 debug 日志, 出現問題時, Rx FIFO full & overrun.
附件 SPI0 驅動 加上清除 FIFO 的控制, 請試一下.
[spi-nuc980-spi0.zip](https://github.com/user-attachments/files/15786688/spi-nuc980-spi0.zip)