OpenNuvoton/NUC980-linux-4.4.y #141: 串口2400波特率使用问题

SoSuperSmall, 2024-09-12:
我当前使用的内核是4.4.289, 0d3ebe8的提交，为了使用串口的1200波特率，我把串口1-9的频率从默认的150MHz改为了50MHz，并且串口的PDMA都是开启的，现在遇到了串口在2400波特率的情况下通信不正常的情况，如下面的图所示。
[image]
对端的串口收发，正常的数据交互

[image]
980上串口的收发，接收数据为异常

测试现象：
1、我们现在的测试是使用了MOXA的串口服务器，使用2400的波特率会出现这种问题。我们尝试了1200、4800、9600和115200的波特率都没有这个问题，通信都是正常的。
2、重启980上的进程，相当于重新打开串口并对串口进行属性设置，通信会恢复正常。
3、把ttyS2的PDMA关掉，ttyS2通信起来也正常，其它没有关掉PDMA的串口使用2400通信还是有此问题。
4、把驱动中串口的频率修改回150MHz，测试2400波特率还是有此问题。
5、尝试把MOXA串口服务器替换掉，直接使用串口对接，这个问题反复测试也没有出现。

因为之前在一个客户现场串口没有开PDMA的情况下出现远端设备丢字节的问题，所以我们现在不能通过关PDMA去解决。
麻烦各位工程师帮忙分析一下问题出来哪里，多谢了。

mjchen1, 2024-09-12:
是不是可以量測一下,兩邊的tx訊號,看一下baudrate的誤差是不是比較大

SoSuperSmall, 2024-09-12:
@mjchen1 感谢您的回复，我们现在是加了485的转换电路，将串口转为485去使用的，刚才我们简单测量了一下两边发送时候的波特率，两边基本上是一致的，一个bit大致都为420us，波特率的误差应该不大。
您看还有哪些方向可以做排查的。

mjchen1, 2024-09-12:
是不是可以在nuc980的rx pin搭上一個監聽設備,
ex:把PC com port上的rx和nuc980 rx搭在一起,
看一下,兩者收到的資料是不是一致

SoSuperSmall, 2024-09-13:
我们往nuc980的rx pin上并了一个ch340的模块，接到pc端看的话数据是正常的，但是980读上来的数据就有误了。

mjchen1, 2024-09-13:
cat /proc/tty/driver/serial
看一下rx收到的資料數量,和tx端送來的資料數量是一致的嗎 ...??

SoSuperSmall, 2024-09-13:
我重新上电比对了一下数据，/pro/tty/driver/serial下面的串口统计里面，使用的串口的rx数据比发送过来的要少，每次发送过来10个字节，第一次看统计是收6字节，第二次看收到13字节。