OpenNuvoton/NUC980-linux-4.4.y #142: SPI NAND可以通过NuWrite下载但是通过SPI NAND启动时启动不了 NAND型号为XT26G04CWSIGA

zyb111, 2024-11-01:
uboot相关配置
[image]

[image]
通过NuWrite下载时，芯片ID识别正确，为0x0b1300   但是在启动时好像是卡在spl里面：
NUC980 IBR 20180813
Boot from SPI-NAND
DDR-OK
finish SPI dow

cwweng-nuvoton, 2024-11-01:
Hi,

Following steps are for XTX SPI NAND boot.
1. Edit include/configs/nuc980_evb.h. Enable definition “CONFIG_SYS_USE_SPIFLASH”.
Attached is my nuc980_evb.h 
[nuc980_evb.zip](https://github.com/user-attachments/files/17597412/nuc980_evb.zip)

2. Enable NUC980 SPI driver
[image]

3. Enable XTX SPI NAND flash Support
[image]

4. Enable sf command
[image]

Then XTX SPI NAND boot successfully.
[image]

My U-Boot binary is as below.
[u-boot.zip](https://github.com/user-attachments/files/17597418/u-boot.zip)

zyb111, 2024-11-01:
我直接使用了您u-boot里面的文件下载到NUC980DK71YC这个芯片里面，并且下载前已经清空了flash里面的内容，启动后依然打印
NUC980 IBR 20180813

Boot from SPI-NAND

DDR-OK

finish SPI dow  有没有可能并不支持这款NAND芯片，或者我们买到了不是正品的芯片

zyb111, 2024-11-01:
您发的图片中的配置我也配置了，结果也是一样的

cwweng-nuvoton, 2024-11-01:
U-Boot 有支持 XTX XT26G04CWSIGA SPI NAND
[image]

zyb111, 2024-11-01:
这个我也看到了我们用的NAND型号全称是XT26G04CWSIGA-B，有没有可能我们这个NAND芯片后面加了-B是特殊版本的NAND芯片导致的

cwweng-nuvoton, 2024-11-01:
那你如果用 NuWriter Download to DDR 的功能, U-Boot 跑得起來嗎 ?
[image]

zyb111, 2024-11-01:
我马上试一下

zyb111, 2024-11-01:
可以的
[image]

cwweng-nuvoton, 2024-11-01:
同樣用 NuWriter Download to DDR 的功能, 可以將 SPL 帶起來嗎 ?
[image]

[image]

cwweng-nuvoton, 2024-11-01:
如果 SPL 也沒問題的話
SPI NAND boot 流程: IBR (boot rom) -> SPL -> Main U-Boot
你那邊 SPL 和 Main U-Boot 都沒問題

但 IBR 帶不起 SPL ?
這不太合理.

zyb111, 2024-11-01:
这个板子之前用过W25N01是可以正常使用的

zyb111, 2024-11-01:
刚才没看到SPL这个操作，试了一下，结果如图示
[image]

zyb111, 2024-11-01:
现在SPL已经运行了起来但是还是没有加载uboot ，是不是可以说明spl确实不支持这款芯片

zyb111, 2024-11-01:
我认为应该是SPL里面不支持这款NAND芯片,我刚才换上了新购买的W25N01可以正常使用

zyb111, 2024-11-04:
您好，按照您的配置下载后还是一样的情况，下载您之前提供的uboot文件夹下的也是一样的结果，今天买到了新的芯片XT26G04CWSIGA不带后缀的，也是一样的结果

zyb111, 2024-11-04:
您好，非常感谢你的回答，现在找到了问题，将Nuwrite更换为1.19版本解决了这个问题，可能1.19这个版本是针对这款芯片是最新的

cwweng-nuvoton, 2024-11-04:
最近一年來, NuWriter 針對 SPI NAND 的改動只有下面
[image]
也就是修正了沒有確實將  "User Defined" 的 Page Size 和 Spare Area 資訊寫入 SPI NAND Flash header.

而 XT26G04CWSIGA SPI NAND 的 page size 是 4096 (非一般常見的 2048)
所以使用舊版 NuWriter 燒錄 XT26G04CWSIGA SPI NAND 的確會有問題.